arendst/Tasmota #680: Sonoff Basic reboot would always trigger Relay ON regardless of options settings

joonhshin, 2017-08-02:
Firmware v 5.2.3
Sonoff Basic

Following options are set:
```
PowerRetain = 1
ButtonRetain = 1
PowerOnState = 0
```

Sonoff Basic would randomly reboot where the "Reboot reason" would state "Power On" (what does this mean anyway?). The relay is OFF when the random reboot occurs, then upon reboot the relay would always be ON.

This is what the console log says:

```
00:00:03 HTTP: Webserver active on sonoff04-0816.local with IP address 192.168.10.104
06:37:20 MQTT: Attempting connection...
06:37:22 mDNS: Query done with 0 mqtt services found
06:37:22 MQTT: Connected
06:37:22 MQTT: tele/sonoff04/LWT = Online (retained)
06:37:22 MQTT: cmnd/sonoff04/POWER = 
06:37:22 MQTT: stat/sonoff04/RESULT = {"POWER":"ON"}
06:37:22 MQTT: stat/sonoff04/POWER = ON (retained)
06:37:22 MQTT: tele/sonoff04/INFO1 = {"Module":"Sonoff Basic", "Version":"5.2.3", "FallbackTopic":"sonoff04", "GroupTopic":"sonoffs"}
06:37:22 MQTT: tele/sonoff04/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff04-0816", "IPaddress":"192.168.10.104"}
06:37:22 MQTT: tele/sonoff04/INFO3 = {"Started":"Power on"}
06:37:22 MQTT: stat/sonoff04/RESULT = {"POWER":"ON"}
06:37:22 MQTT: stat/sonoff04/POWER = ON (retained)
06:37:23 MQTT: stat/sonoff04/RESULT = {"POWER":"ON"}
06:37:23 MQTT: stat/sonoff04/POWER = ON (retained)
06:47:16 APP: Serial logging disabled
```

davidelang, 2017-08-03:
it looks to me like your mqtt server is issiing the command to turn on the 
power, the lst command it remembers seeing is to turn on the device, so when it 
comes up, it sets it back to what it thinks is correct.
try disabling the retain settings (there's a wiki page on how to convince the 
mqtt server to forget the retained settings)

rebooting with Power On means that the sonoff thinks it lost power.

joonhshin, 2017-08-03:
Thanks for the comment. What puzzles me is that the last command sent by MQTT to Sonoff was to turn "OFF". So I'd assume MQTT would send the last "OFF" command again upon Sonoff rebooting.

Besides, the "cmnd" topic being sent and received seem to have no payload. Any ideas as to why this would still trigger an "ON" behavior?

`06:37:22 MQTT: cmnd/sonoff04/POWER = 
`

mkh595, 2017-08-03:
I had same problem and it was caused by retained message on MQTT broker.
You can open MQTT client, subscribe for stat/# and you will get all retained messages.

my log now looks:

`00:00:10 MQTT: home/tele/home203/LWT = Online (retained)
00:00:10 MQTT: home/cmnd/home203/POWER = 
00:00:10 MQTT: home/stat/home203/RESULT = {"POWER":"OFF"}
00:00:10 MQTT: home/stat/home203/POWER = OFF

00:00:11 MQTT: home/stat/home203/RESULT = {"POWER":"OFF"}
00:00:11 MQTT: home/stat/home203/POWER = OFF``

but your shows retained flag:

06:37:22 MQTT: cmnd/sonoff04/POWER = 
06:37:22 MQTT: stat/sonoff04/RESULT = {"POWER":"ON"}
06:37:22 MQTT: stat/sonoff04/POWER = ON **(retained)**

closed 2017-08-04 by joonhshin