arendst/Tasmota #681: Tasmota on Wemos D1 mini

gheesung, 2017-08-02:
I have successfully run Tasmota ver 5.3 on Wemos D1 Mini. Recently I have upgraded to v5.5 and encountered the errors and the following is from the serial output. I tried to revert back to 5.3 and the having the same errors. Do anyone have the same problem as me?

`
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

00:00:00 APP: Project brownstone Sonoff (Topic livingrm/temp, Fallback DVES_3AA0EB, GroupTopic sonoffs) Version 5.5.0
00:00:02 Wifi: Connected
00:00:02 mDNS: Initialized
00:00:02 HTTP: Webserver active on livingrm/temp-0235.local with IP address 192.168.0.96
00:00:02 UPnP: Multicast (re)joined
00:00:07 UPnP: Multicast (re)joined
00:00:13 MQTT: Attempting connection...
00:00:14 mDNS: Query done with 0 mqtt services found
00:00:18 MQTT: Connected
00:00:18 MQTT: ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ = Online (retained)
00:00:18 MQTT: ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ = 
00:00:18 MQTT: ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ = {"Module":"WeMos D1 mini", "Version":"5.5.0", "FallbackTopic":"DVES_3AA0EB", "GroupTopic":"sonoffs"}
00:00:18 MQTT: ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ = {"WebserverMode":"Admin", "Hostname":"livingrm/temp-0235", "IPaddress":"192.168.0.96"}
00:00:18 MQTT: ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ = {"Started":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x402261dc epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000001 depc:0x00

Exception (28):
epc1=0x402261dc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000001 depc=0x00000000

ctx: cont 
sp: 3fff2200 end: 3fff2780 offset: 01a0

>>>stack>>>
3fff23a0:  3fff08cb 00000001 00000000 402124f9  
3fff23b0:  ffffffff ffffffff ffffffff ffffffff  
3fff23c0:  696c6e4f 4f00656e 00000033 4010020c  
3fff23d0:  544c5352 6552203a 76696563 6f742065  
3fff23e0:  20636970 ffffffff ffffffff ffffffff  
3fff23f0:  ffffffff ffffffff ffffffff ffffffff  
3fff2400:  ffffffff ffffffff ffffffff ffffffff  
3fff2410:  ffffffff ffffffff ffffffff ffffffff  
3fff2420:  ffffffff ffffffff ffffffff ffffffff  
3fff2430:  ffffffff ffffffff ffffffff ffffffff  
3fff2440:  ffffffff 2cffffff 74616420 69732061  
3fff2450:  3620657a 6164202c 4f206174 6e696c6e  
3fff2460:  70650065 303a3363 30303078 30303030  
3fff2470:  78652030 64617663 303a7264 30303078  
3fff2480:  30303030 65642031 303a6370 00303078  
3fff2490:  ffffffff ffffffff ffffffff ffffffff  
3fff24a0:  ffffffff ffffffff ffffffff ffffffff  
3fff24b0:  ffffffff ffffffff ffffffff ffffffff  
3fff24c0:  ffffffff ffffffff ffffffff ffffffff  
3fff24d0:  ffffffff ffffffff ffffffff ffffffff  
3fff24e0:  ffffffff ffffffff ffffffff ffffffff  
3fff24f0:  00ffffff 00000000 00000000 00000000  
3fff2500:  00000000 00000000 00000000 00000000  
3fff2510:  00000000 00000000 3fff2634 402099fa  
3fff2520:  3fff5600 000004be 3fff36ec 00000080  
3fff2530:  3fff10a4 00000001 3fff25d0 4020f77a  
3fff2540:  00000001 3fff25d0 3fff36ec 40223eaa  
3fff2550:  3fff25a0 3fff2590 0000000c 7fffffff  
3fff2560:  3fff3094 00000000 3fff550c 0000006e  
3fff2570:  00000003 3fff2620 00000003 40223ed8  
3fff2580:  4024cb8e 00000003 3fff2620 00000001  
3fff2590:  4024dcb3 6e696f6a 3f006465 00000000  
3fff25a0:  3fff317c 4021a20c 3fff36ec 40223eaa  
3fff25b0:  3fff2600 3fff25f0 0000000c 402098ce  
3fff25c0:  00000002 3fff2630 3fff4164 0000006e  
3fff25d0:  696c6e4f 0000656e 00000003 40223ed8  
3fff25e0:  3fff2600 303a3030 33313a30 00000000  
3fff25f0:  40103109 00040000 7fffffff 00000000  
3fff2600:  00000022 40103106 00040000 0000006e  
3fff2610:  3ffeebc8 401020ea 00000003 4020d57c  
3fff2620:  506e5055 ffffffff ffffffff ffffffff  
3fff2630:  ffffffff ffffffff ffffffff ffffffff  
3fff2640:  ffffffff ffffffff ffffffff ffffffff  
3fff2650:  ffffffff ffffffff ffffffff ffffffff  
3fff2660:  ffffffff ffffffff ffffffff ffffffff  
3fff2670:  ffffffff ffffffff ffffffff ffffffff  
3fff2680:  ffffffff 00ffffff 0000006c 000002df  
3fff2690:  0000480f 3fff08cb 00000063 4021afee  
3fff26a0:  4024cb8e 3fff26e2 00000065 4021afee  
3fff26b0:  3ffe86fc 3ffe8704 3fff08c4 40218dac  
3fff26c0:  00000001 00000003 00000006 3fff23c0  
3fff26d0:  00000068 00000000 3fff23c0 40218ed5  
3fff26e0:  6965006d 696c2063 3fff42b4 4021b38e  
3fff26f0:  3fff2720 00000001 00000063 00000001  
3fff2700:  00000063 00004909 3fff08c4 40223f7c  
3fff2710:  3fffdad0 00004909 3fff08c4 40219485  
3fff2720:  3ffe9601 00000000 00001388 3fff163c  
3fff2730:  00000004 3fff08c7 0000006d 3fff1760  
3fff2740:  3fffdad0 00000000 3fff1748 3fff1750  
3fff2750:  3fffdad0 00000000 3fff0ea4 40218be4  
3fff2760:  3fffdad0 00000000 3fff1748 40223bd0  
3fff2770:  feefeffe feefeffe 3fff1760 4010070c  
<<<stack<<<
`

arendst, 2017-08-02:
Perform a ``reset 1`` to reset your configuration.

gheesung, 2017-08-02:
I can't even go into the web page. It keeps on prompting me for the password which I did not set.

How do I do a reset 1?

arendst, 2017-08-02:
Your wemos should have a serial interface. You have 18 seconds before it reboots. As soon as it is rebooted execute command ``seriallog 3``.

Then after the next reboot execute command ``cfgdump 0 16``

gheesung, 2017-08-02:
It is working now. I have set `#define CFG_HOLDER` to an arbitrary value, compiled and uploaded the sketch again. The serial output is correct and I can go into the web console.

I am not sure what is happening and when possible,  I will like to know what is the exact cause.

arendst, 2017-08-02:
Me too!

That's why I asked you to perform some serial commands to get me information for investigation. 

As the configuration is now overwritten we are done here.

gheesung, 2017-08-02:
Oh! By the time I read your message, I have already recompiled the sketch. Next time if I encountered it again, I will try to dump it. 

Thanks for your help and the great software. I am going to flash it to all my Sonoff devices.

closed 2017-08-02 by gheesung