arendst/Tasmota #683: No serial data on Sonoff Basic

mocoloco69, 2017-08-03:
I just bought 2 new Sonoff switches and flashed the TASMOTA firmware to them and after that nothing happens. I don't get any flashing LED, no data on the serial port, no clicking from the relay when the button is pushed, nothing. The Arduino IDE says that the firmware uploaded correctly, but nothing seems to be happening. I have already done this with another Sonoff Basic, and it works flawlessly. I am using the same FTDI connector, the same default firmware (with no modifications), everything is the same as I did it before, but I'm not getting any response from the Sonoff. Before I tried to upload the firmware, the light would come on and blink on one Sonoff (I forgot to check the other one before I flashed it), but since I flashed it, nothing happens. Did I miss something? Did I somehow manage to "brick" both of my Sonoff switches? Any help would be greatly appreciated. I'm not sure where to start in troubleshooting this issue.

ionciubotaru, 2017-08-03:
Try this https://github.com/arendst/Sonoff-Tasmota/issues/636#issuecomment-318404888

mocoloco69, 2017-08-03:
ionciubotaru: I need something that will run on Linux Mint, not Windoze. I'm using the Arduino IDE, will this work with the Arduino IDE, or do I have to use the ESP Flash Tool?

ionciubotaru, 2017-08-03:
Use `$./esptool -cd ck -cb 115200 -cp COM97 -ca 0x00000 -cf ./s530.ino.bin`

mocoloco69, 2017-08-03:
Where do I get the esptool file and the s530.ino.bin file? And why is it using COM97? I'm new to a lot of this. If I know where to get the files, I can follow your instructions. Thank you for taking the time to help me.

mocoloco69, 2017-08-03:
So, I found out that the Arduino IDE has an EEPROM clear program under the examples for the Generic ESP8266 Module. It is only for 512 bytes of eeprom. I set it to 1024 bytes, uploaded it, and then uploaded the TASMOTA firmware, afterwhich I got this in the serial monitor:

hksum 0xef
csum 0xef
csum err
ets_main.c 

It looks like it is cut off on the left side, but that is all that was on the screen.

ionciubotaru, 2017-08-04:
You are totally confused. The eeprom you refer if just a small part of the physical eeprom.
The bin file is generated by the arduino compiler. 
COM97 if the FTDI com port
esptool is available on expressif website

mocoloco69, 2017-08-04:
You're right, I am totally confused. Like I said, I am new to all of this. I have one Sonoff switch up and running with no problems at all which is connected to my swampcooler and a WeMos connected to my thermostat that turns my swampcooler on and off based on the temperature. So I understand a little bit about how this is supposed to work. However, it isn't working, which is why I'm so confused. I don't know the proper names of everything, I'm trying to figure all this out as I go. I connected my FTDI adapter to my working Sonoff and I get messages on the serial port, the Sonoff functions flawlessly as would be expected. So, my FTDI adapter is working properly. Other threads I have read have said sometimes the FTDI adapters don't supply enough power for proper operation. I have now ruled this out.

I understand that the .bin file is generated by the Arduino IDE, but the instructions you gave me for the command prompt:

Use $./esptool -cd ck -cb 115200 -cp COM97 -ca 0x00000 -cf ./s530.ino.bin

call up the esptool (which I don't have installed on my linux machine) and also a file named s530.ino.bin which I also don't have. My FTDI port is /dev/ttyUSB0 which is why I was confused to see a COM97 in your instructions. I'm not trying to be difficult, I'm just trying to understand all the different options so I can change the ones I need to change to match my laptop (like perhaps changing COM97 to /dev/ttyUSB0). If I knew how to install esptool for Linux, and where to get the s530.ino file, I would follow your instructions. As I don't have this information at the moment, I'm trying a lot of different things with absolutely no success. I've went to the expressif website and don't find anything to install the ESP Flash Tool for Linux. It is only for Windows.

ionciubotaru, 2017-08-04:
I gave you the instructions because you said you don't like windows.
Linux is not for beginners at this level. Use Arduino for Windows step by step and you will succed. After that you can try linux

mocoloco69, 2017-08-04:
I've been using Linux for years. I'm not a complete newbie, but I am a noob with Sonoff and Arduino. I have one Sonoff that works, 2 that don't. How do I install the esptool for Linux?

mocoloco69, 2017-08-04:
OK, for anyone who wants to know how to install the ESP Flash Tool (esptool) on a Linux machine, and can't seem to get any help in doing so, like I couldn't. Here is how I was able to install it:

First you have to install "pip" for python:

sudo apt install python-pip

Next, you need to upgrade pip to the latest version. I don't know why it doesn't install the latest version, but it told me I needed to upgrade:

sudo -H pip install --upgrade pip

make sure you put the -H after sudo, or you will get an error.

Then, you need to install setup tools:

sudo -H pip install setuptools

Now, we can install ESPTOOL:

sudo -H pip install esptool

Once this is done, you can use ESPTOOL from the command line. However, DO NOT use the command in the post above, IT DOESN'T WORK! For more information on using ESPTOOL see the read me: https://github.com/espressif/esptool/blob/master/README.md.

Here is the command I used to erase my flash:

esptool.py -b 115200 -p /dev/ttyUSB0 erase_flash

The "-b" option is for your baud rate. If your baud rate is different, change it. The "-p" option is the port your FTDI adapter is connected to. If you are using Windoze, it may be a COM port. Change it to whatever port you use for the Arduino IDE to upload sketches. The "erase_flash" option is self explanatory and couldn't be simpler. You do not need a file to upload. 

Here is the result I got of typing in the above command:

esptool.py v2.0.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.5s
Hard resetting...
 
So, everything worked as far as that went, supposedly... However, after I upload the TASMOTA firmware to the Sonoff, I have exactly the same problem that I had before. NOTHING HAPPENS! I don't get any activity on the serial port, no LED flashes, nothing. I'd even be OK if the magic smoke escaped! At least that would be something! But, nothing. So, it supposedly erased my flash. 3.5 seconds seems awfully fast. I think my Sonoffs are just junk. Next time I'll buy them directly from iTead, and make sure they work BEFORE I flash them. Oh well. At least they were only $6 each.

Anyone else have any ideas of something I can try?

arendst, 2017-08-04:
What firmware are you uploading? The provided one from the latest release or your own compiled version?

mocoloco69, 2017-08-04:
arendst: Thank you for commenting. I've been trying to do 2 different things: first, I've been trying to upload the TASMOTA software to a Sonoff Basic using the Arduino IDE on Linux, which I have successfully done before just 2 weeks ago. My new Sonoff Basic switches (2 of them) don't seem to be working. It has been suggested that I erase the flash to make sure the TASMOTA firmware is uploaded to a "clean" flash memory. So, the second thing I was trying to do, was to erase my flash memory using the esptool on Linux. Once I figured out how to install it, everything worked fine. I'm not using the esptool to upload any files, nor do I want to. I was only using it to erase the flash. Everytime I upload firmware to the Sonoff Basic, it seems to work without any errors at all. However, I go no response from the Sonoff. No LED activity, no serial port activity, nothing. I have verified that my FTDI adapter is working just fine by connecting it to my working Sonoff Basic while it was disconnected from Mains voltage and everything worked. I got LED activity, I was able to toggle the relay, and I was able to see information on the serial port. That Sonoff Basic has the TAMOTA firmware installed also. I am using the same exact firmware, with no modifications, just as it was downloaded about 2 weeks ago. I am using the same Arduino IDE setup also. None of the settings have been changed at all. I double checked everything just to make sure. I think my Sonoff switches are bad.

arendst, 2017-08-04:
Whats the flashmode setting in your arduino ide?

mocoloco69, 2017-08-04:
DIO

arendst, 2017-08-04:
Right. Now change that to DOUT. Recompile, upload and enjoy your sonoff basic.

mocoloco69, 2017-08-04:
HOLY CRAP!!! WTF?? IT WORKS!! THANK YOU SO MUCH!! Now, why? The flash mode has ALWAYS been set to DIO, and it worked before... Oh well. At this point, I don't care. It is working great!! Thank you so much Theo!! You are awesome!

arendst, 2017-08-04:
I guess I have a problem trying to explain the situation clearly to people...

In my earlier answer I tried to explain the situation regarding esp8266 and esp8285...

For anyone reading this:

STOP USING DIO/QIO AND FINALLY START USING DOUT ONLY AS PERFECTLY DOCUMENTED IN THE WIKI

mocoloco69, 2017-08-05:
My bad. When you mentioned the DOUT setting, I didn't know that you were talking about the setting in the Arduino IDE. You gave a command line command, and I thought the DOUT was specific to the command line interface. I misunderstood. I really apologize. I'm new to the world of Sonoff and Arduino and don't understand a lot of the settings, but I am trying to learn. When I checked my settings, I only checked them against what I knew I had used before. I didn't check the wiki or anything else. I was also confused between the settings for esptool on the command line, and the Arduino IDE. Now that I am able to compare them, I can see the similarities. You did fine explaining. I did poorly understanding. Thank you so much for taking the time to explain the situation and fix my problem. I really appreciate it.

closed 2017-08-05 by arendst