arendst/Tasmota #684: Reset boot count (request)

mvveelen, 2017-08-03:
Hi,

by setting up the Tasmota firmware, the device needs at least 10 to 15 reboots by saving the settings from every single page. Also, when testing the setup there are probably some reboots needed.

Now, I would like to keep an eye on the devices for spontaneous reboots or such and therefor I'd like to reset the boot count to zero. 

Would it be possible to add such an option ?

stefanbode, 2017-08-03:
If you use MQTT you can see on the change of LWT the online/offline changes on every reboot. You get on INFO3 topic the reason for the reboot and count these and make evaluations. All these methods are quite superior just than resetting the bootcount to 0. Without MQTT the evaluation is quite complex

mvveelen, 2017-08-04:
Well, I'm (trying to) use MQTT with Domoticz, but how does this relate to starting with a boot count of zero? A message / signal when a reboot occurs would be a nice extra (wouldn't even know how to get this in Domoticz, but OK), but I simply would like to start from zero :)

Some reboots are caused by a 'hardware watchdog', but then I still don't know what happened. Right?

arendst, 2017-08-05:
As @stefanbode already noted after every reboot you'll receive three MQTT messages of which INFO3 reports the reboot reason. You could then execute command ``Status 1`` to retrieve your required Bootcount.

mkh595, 2017-08-06:
But requirement was to reset boot counter to 0. 
@mvveelen: When you re-flash firmware, counter will be reseted. But you have to restore also last configuration.

mvveelen, 2018-01-29:
Reflashing isn't an option. I would like to be able to reset it manually, but if this isn't going to be added, I'll leave it.

closed 2018-06-13 by stale[bot]