arendst/Tasmota #686: Enhancement request: Stabelize AnalogRead

stefanbode, 2017-08-03:
Hi,

I use quite a lot of analog sensors and found out that the values quite vary a lot between the different measures (Typical +- 15%). Also from time to time, there is an "invalid" value reported. 

I found to get rid of the problem to do 100 reads with a delay(1) in between and use the average. Now my values are quite more stable from measure to measure. Therefore I vote for doing some average instead of just one read. Maybe 20 reads are fine, but I do not see any timing issues with the 100 iterations.

davidelang, 2017-08-03:
100 reads with delay(1) between them would make the device unresponsive for 100 
seconds at a time, so some other method would be needed to accumulate and 
average the readings :-)

stefanbode, 2017-08-04:
@davidelang : no, the delay is Milliseconds. Therefore it takes 0.1+ seconds for 100 readings. I assume it should be done in less than 0.2 seconds including the average calculation and A/D conversion.

PeggyFree, 2017-08-04:
You have to use a timer for your need. See
http://www.switchdoc.com/2015/10/iot-esp8266-timer-tutorial-arduino-ide/

stefanbode, 2017-08-04:
@PeggyFree In general I agree, but for a task lasting only 0.2 seconds, there is no known problems with Wifi and MQTT visible. The timer stuff is afaik only to avoid blocking the device and introduce a reset because of "blocked loop".

arendst, 2017-08-05:
Will add averaging 32 attempts  in next release.

arendst, 2017-08-09:
@stefanbode does the fix work for you?

stefanbode, 2017-08-17:
Works great!

closed 2017-08-17 by stefanbode