arendst/Tasmota #687: How to send command via Grouptopics and fallback topics

gheesung, 2017-08-04:
For normal switching on/off one switch, the following commands worked.

`mosquitto_pub -h openhabianpi -t brownstone/livingrm/switch01/cmnd/POWER -m 0`

My GroupTopic is `livingrmswitch` and I am trying to switch on/off all the switches in the living room. May I know how to write the command correctly? Currently, I tried the following but it doesn't work. 
`mosquitto_pub -h openhabianpi -t livingrmswitch/cmnd/POWER -m 0`

Thanks

davidelang, 2017-08-04:
On Thu, 3 Aug 2017, gheesung wrote:

> For normal switching on/off one switch, the following commands worked.
>
> `mosquitto_pub -h openhabianpi -t brownstone/livingrm/switch01/cmnd/POWER -m 0`

for this to work, you have changed the format of the topics, what is the device 
topic in this string? whatever that is, change it to the grouptopic.

gheesung, 2017-08-04:
great, it works for grouptopic.

my device topic is `livingrm/switch01` and I changed it to `mosquitto_pub -h openhabianpi -t brownstone/livingrmswitch/cmnd/POWER -m 1`

How about Fallback topics? My current device fallback is `DVES_F90887`

davidelang, 2017-08-04:
all topic options work the same way.

gheesung, 2017-08-04:
I have tried `mosquitto_pub -h openhabianpi -t brownstone/DVES_F90887/cmnd/POWER -m 0` but the switch did not turn off.  Is there anything I may have entered wrongly?

closed 2017-08-05 by gheesung