arendst/Tasmota #688: Sonoff POW: Send tele data based on Current or Power Threshold

siam28, 2017-08-04:
Similar to the OP who posted this [issue](https://github.com/arendst/Sonoff-Tasmota/issues/349), I believe a feature that sends tele data based on a current or power threshold would be beneficial to avoid flooding an MQTT broker with unnecessary tele data when the tele period is required to be small to catch device state updates.
Z-wave switches tend to have this feature so it would be nice to bring it to Sonoff. 

Note that this would still require doing the logic to determine device state (on/off, speed, etc.) in a home automation software/hub. This feature would just make getting tele data more efficient.

Are there any plans to integrate this feature?

mkh595, 2017-08-04:
I would be also glad to have 'dynamic' telemetry period. Not only for POW where it is clear. I use for them period 10 sec but when relay is off, all measures are 0 and it has no sense to send it each 10 seconds.
It would be nice also for temperature sensors which send data to other units which use them for control, As was stated above, if there is no change in value, then is no reason to send same data so often.

davidelang, 2017-08-04:
note that the sensor data is only collected every teleperiod seconds, so it's 
not that you are gathering the data continually and only sending it every X 
seconds, it's that you are only gathering the data every X seconds.

Tasmota doesn't gather the data into variables and then format it for the mqtt 
message, it executes a function in each module that returns a string that's 
added to the tele output. you could look at remembering the string from last 
time and not sending it if it hasn't changed, but that will double the memory 
you need for that string, and we are tight enough on ram that you could run into 
grief doing this.

siam28, 2017-08-05:
Thanks for the clarifications. So setting the teleperiod value to a small value to catch device state changes is unavoidable. I still think there is value in updating the MQTT topic only when there are changes above some threshold. 

Regarding the memory issue, we could make the threshold-ing logic disabled by default and let the user enable it if their device has enough memory. Another option is to only threshold current or power (not both).

arendst, 2017-08-05:
Adding Energy tele message to margin change is implemented in next release.

mkh595, 2017-08-07:
How it should work? I downloaded version 5.5.1 but telemetry data are send still according to telemetry period, not according to their change.

arendst, 2017-08-07:
Look in wiki commands Sonoff pow specific and set current, voltage and power thresholds. If threshold is met tele data is send too.

mkh595, 2017-08-08:
I set CurrentHigh to 10 mA and expect, that message will be send for each measurement (like continuous updated in Web gui)  because current 'Current' 0.367 is always abow threshold.
But message is sent in interval more than 1:10 minutes.
TelePeriod is set to 60 seconds.
I assume that only one threshold should be met to send message, not all (current, voltage, power) correct?
Also message `MARGINS = {"CurrentHigh":"ON"}` was sent only once. Could be problem there that was recognized only first trigger?

09:48:42 CMND: cmnd/CurrentHigh 10
09:48:42 MQTT: home/stat/home214/RESULT = {"CurrentHigh":"10"}
09:48:43 MQTT: home/tele/home214/MARGINS = {"CurrentHigh":"ON"}
09:48:43 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:48:43", "Total":1.300, "Yesterday":0.494, "Today":0.806, "Period":1, "Power":82, "Factor":1.00, "Voltage":223, "Current":0.367}
09:48:52 MQTT: home/tele/home214/STATE = {"Time":"2017-08-08T09:48:52", 
09:48:52 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:48:52",
09:50:04 MQTT: home/tele/home214/STATE = {"Time":"2017-08-08T09:50:04", 
09:50:04 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:50:04", 
09:51:15 MQTT: home/tele/home214/STATE = {"Time":"2017-08-08T09:51:15", 
09:51:15 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:51:15", 
09:51:46 CMND: cmnd/CurrentHigh
09:51:46 MQTT: home/stat/home214/RESULT = {"CurrentHigh":"10"}
09:52:25 MQTT: home/tele/home214/STATE = {"Time":"2017-08-08T09:52:25", 
09:52:25 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:52:25", 
09:53:34 MQTT: home/tele/home214/STATE = {"Time":"2017-08-08T09:53:34", 
09:53:34 MQTT: home/tele/home214/ENERGY = {"Time":"2017-08-08T09:53:34",

mkh595, 2017-08-08:
I set also limits for `RESULT = {"PowerHigh":"5"}` and `RESULT = ["VoltageHigh":"10"}` and `MARGINS = {"PowerHigh":"ON"}` and `MARGINS = {"VoltageHigh":"ON"}` are triggered only once

carryonrewardless, 2018-01-30:
Sorry to re-open an old conversation. @mkh595 you need to set Teleperiod to zero. Then it will give one extra message after the value goes below the threshold and then stops sending until the value goes above the threshold again.

jshank, 2018-02-13:
I'm actually looking more for a trigger. If `CurrentHigh 10000` (10amps), the wash spin cycle is running and I get a sweet `tele/sonoff4/MARGINS {"CurrentHigh":"ON"}`. When the spin cycle is done, `CurrentLow 1` (1ma) is the second trigger that tells me the wash is done.

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

iliyang, 2020-04-29:
I found this thread while looking for adaptive telemetry intervals but for a temperature sensor. (I have one connected to a Shelly relay that controls an electric heater.)

It makes a lot of sense to only send a new MQTT temperature reading when the temperature has changed "significantly" since the last MQTT message. I have a Xiaomi Zigbee sensor that does exactly this, which seems to strike a good balance between battery usage and accuracy. Here the concern is about MQTT flooding (not about power usage), but the argument is the same.

Is such a thing now possible with Tasmota? (Or ESPHome, if you don't mind me asking?)

ascillato, 2020-04-29:
I don't know about ESPHome, but it is possible with Tasmota. It is your lucky day because exactly that use case is already explained with examples in the docs at:
https://tasmota.github.io/docs/Rules/#transmit-sensor-value-only-when-a-delta-is-reached

iliyang, 2020-04-29:
But of course, with a rule! Thanks a lot @ascillato!