arendst/Tasmota #689: changing gpio button for - Button usage - to button 3 intead 1 or GPIO0

roblad, 2017-08-04:
Hi,

I prepared my own device based on SUPLA 
https://github.com/SUPLA 
and that project to 
https://forum.supla.org/download/file.php?id=171. 
See also that 
https://www.wykop.pl/link/3325399/diy-supla-do-puszki-instalacyjnej-podtynkowej-supla-org/

  I have 3 buttons gpio14 gpio12 gpio4 and additionally I have gpio0 for flashing pinned. I would like change Button usage to be assigned to gpio4  by default for operations. Where in the source code I can change it before compilation. I guess that it is somewhere in core_esp8266_wiring_digital.c. I need PINmode change from 0 to 4 for configuration usage assigned to "Button usage" functions. Is it possible ?

Thanks

davidelang, 2017-08-04:
see sonoff_template.ino and the module types defined at the bottom of the file 
(you can use the wemos module and define everything in the config)
k

roblad, 2017-08-04:
Hi,

There is sonoff_template.h and there no definition of GPIO. I need tu use gpio14 as Button1 gpio12 Button2 and gpio4 as Button3 but I would like to assign functionality to Button3  as it is described "Button usage"

davidelang, 2017-08-04:
look at the module definitions at the bottom of the file, that assigns the 
default function for every GPIO (plus the analog pin)

look at how the existing modules are defined.
k

closed 2018-05-07 by stale[bot]