arendst/Tasmota #690: Inconsistent Property Names in Messages

DeanCording, 2017-08-04:
I've noticed a few messages where the property names are inconsistent when applied to the same values.

For example, INFO2 and STATUS5

`tele/lower/INFO2 {"WebserverMode":"Admin", "Hostname":"lower-6111", "IPaddress":"192.168.1.1"}`

`stat/lower/STATUS5 {"StatusNET":{"Host":"lower-6111", "IP":"192.168.1.1", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"5C:CF:7F:F9:77:DF", "Webserver":2, "WifiConfig":2}}`

INFO2 returns the values _Hostname_ and _IPaddress_, whereas STATUS5 returns _Host_ and _IP_ for exactly the same values.

The Wifi commands use _Hostname_ and _IPaddress1_ for the same values.

arendst, 2017-08-04:
Another 11 bytes from my code space.... ;-)

StatusNET will be fixed in next release.

davidelang, 2017-08-04:
or another 11 bytes saved if you change the other output :-)
k

DeanCording, 2017-08-06:
Confirmed fixed.

closed 2017-08-06 by DeanCording