arendst/Tasmota #692: After configuring the modele it doesn't restart but hangs in wdt reset

pascalsaul, 2017-08-05:
Flashed the new 5.5.1 firmware with own user_config.h and when going to the webinterface I first change the module. 

In this case from Basic to Touch:
`18:57:05 HTTP: Sonoff Touch Module, GPIO1 0, GPIO3 0
18:57:06 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset`

It happens with every firmware and every module as far as I tested it :)

mkh595, 2017-08-06:
I guess it is related to ESP and not to Tasmota firmware because  the same happen to me also for other arduino codes and also for Wemos or NodeMCU, not only Sonoffs

DeanCording, 2017-08-06:
I've noticed this with a bunch of Sonoff devices I've reflashed from the original Sonoff firmware using versions from 5.0 on.  Happens with all devices - Basic, TH, Pow, 4 Channel.

Only happens on first change of module - from default Basic to actual.  Subsequent changes to module type do not hang.

Devices work ok after a power cycle.

pascalsaul, 2017-08-06:
Yes, I noticed the same behaviour DeanCording. It would be nice if it can fixed for the good user experience :)

arendst, 2017-08-06:
https://github.com/esp8266/Arduino/tree/master/doc/faq#i-have-observed-a-case-when-esprestart-doesnt-work-what-is-the-reason-for-that

pascalsaul, 2017-08-08:
Ai, what a nice platform bug. Then we'll need to life with it and update our "add new device" process accordingly.

closed 2017-08-08 by pascalsaul