arendst/Tasmota #693: Configure settings with only one reboot

pascalsaul, 2017-08-05:
For every device I add I need to configure some settings which I can't do through user_config.h.

Main settings I need to adjust are only the names used in different parts in the configuration:
1. Configure Wifi => Hostname
2. Configure MQTT => Client Id
3. Configure MQTT => Topic 
4. Configure Other => Friendly Name 1

I would prefer to do this under Configure Module where there is a field ' Module name'.

During the first setup I can change the "Module type" together with the 'Module name' and press Save.

The 'Module name' would be used for the four bullets above but can also be used when you want to rename the device quickly ;)

In the end you'll save 3 reboots and it will be more user friendly!

arendst, 2017-08-05:
Have a look at the backlog command.

pascalsaul, 2017-08-06:
Ah, didn't see that option yet and it is doing the job, sorry.
mosquitto_pub -t cmnd/basic02/Backlog -m 'FriendlyName basic03;Hostname basic03;MqttClient basic03;Topic basic03'

I still do think it is a nice to have feature for people who aren't really familiar with the command line and doing everything throughout the GUI. Yes, I know people who are playing around with domotica and flashing and avoiding to use the CLI ;) For those it will be a great feature to add. It is up to you to decide to implement this tiny little feature request then.

arendst, 2017-08-06:
GUI is code memory hungry. Webserver is nice to have. MQTT is core business.

pascalsaul, 2017-08-08:
I agree what you're saying but maybe if you want to reach an even bigger public with this nice piece of software ;)

In that case: won't fix

Through MQTT you can do it like:
mosquitto_pub -t cmnd/DVES_8546BB/Backlog -m 'Module 10;FriendlyName touch01;Hostname touch01;MqttClient touch01;Topic touch01'