arendst/Tasmota #695: OTA Update Doesn't Handle HTTP_UPDATE_NO_UPDATES Result Properly

DeanCording, 2017-08-06:
An update server is able to return a HTTP 304 (Not Modified) response code to indicate that there is no update available.  In ESP8266 core version 2.4.0rc1, the OTA http request includes to MD5 hash of the current firmware, which the server can compare with the firmware it has.  If they match, the server returns a HTTP 304 response to indicate that an update is not required.

The ESP8266 httpUpdate library returns HTTP_NO_UPDATES when it receives a HTTP 304 response.  Sonoff-Tasmota only checks for a HTTP_UPDATE_OK and considers any other result as a fail.  As a result, it repeatedly requests the update even though none is available.

Offending code is in sonoff.ino (line 2391)

'otaok = (HTTP_UPDATE_OK == ESPhttpUpdate.update(sysCfg.otaUrl));'

Suggest replacing it with

'otaok = (HTTP_UPDATE_FAILED != ESPhttpUpdate.update(sysCfg.otaUrl));

(Didn't cost any bytes this time :)

davidelang, 2017-08-06:
Is that the only failure message that can happen with this library?

DeanCording, 2017-08-06:
Yes, all internal and external failures, including internal failures such as firmware to big or corrupt header, are mapped onto HTTP_UPDATE_FAILED.

arendst, 2017-08-06:
How about 2.3.0?

DeanCording, 2017-08-07:
Yes, it applies to 2.3.0 as well. Although the client won't send the MD5 hash in the request, the server may have other ways of checking if an update is required and return a 304 if it is not.

arendst, 2017-08-08:
In next release.

closed 2017-08-09 by arendst