arendst/Tasmota #698: HUE App wants to update emulated HUE devices

chinds185a, 2017-08-06:
So I successfully flashed this firmware onto a 4CH sonoff  today. I have turned on the hue emulation. I want to control these devices with Alex and ultimately Apples home kit.

Alex does not discover the devices so I guess I need to add them in the HUE app first? 

Going to the HUE app it find the new bridge and the 4 named devices. However I cannot save the light/add them to a room etc as the HUE app is continuously trying to update the firmware on the new emulated hue bridge and obviously fails.

How can I get each ch discover-able by Alexa? Using wemo emulation works but again only for a single Chanel.

arendst, 2017-08-08:
Update emulated HUE devices as designed. Don't know how to fix. Probably have to tell the App I have a way to new version....

Alexa should work without any Hue App.

Courty40, 2017-08-11:
I've just tried the same thing with a 4ch and version 5.5.2 and I'm seeing the same behaviour.
Alexa see ch1 of the 4ch device in Wemo mode but doesn't detect the Sonoff at all in Hue mode. Tried rebooting/power off etc
I had this working a few versions ago on a 2ch Sonoff so I do t believe it's the Alexa

Courty40, 2017-08-11:
And I can confirm the hue app finds the Sonoff and tries to upgrade it even though the Alexa device search doesn't
Something changed in the Hue discovery protocol maybe ?

arendst, 2017-08-11:
My Hue code changed between version 5.2.4 and 5.3.0 and might be causing you trouble. I notice the same behaviour on my Hue App regarding the firmware update. I just ignore it.

I do not have Alexa so I can't test. If others have problems with it too please let me know.

Courty40, 2017-08-11:
Just remoted into one I know works with Hueand that's a BN-SZ LED and its running 5.3.0

Courty40, 2017-08-11:
More info, Alexa does poll the unit and the unit is responding but Alexa doesn't accept the Sonoff

23:44:31 HTTP: Root
23:44:33 HTTP: Console
23:44:46 Wifi: Checking connection...
23:44:46 Wifi: Connected
23:44:48 UPnP: HUE 3 response packets sent to 192.168.1.16:50000
23:44:48 UPnP: HUE 3 response packets sent to 192.168.1.16:50000

chinds185a, 2017-08-13:
Still having problems with this. Alexa just doesn’t discover the hue devices. 

Is it worth downgrading the firmware used?

altelch, 2017-08-13:
Hi,

would be interesting to see what was changed between the last working version and the current version. I still get the emulation working but discovery mostly takes 3 to 4 tries before it works. Maybe also something was changed in Echo firmware during the last versions. Didn't look into this. Hopefully I find some time the next days to check into this issue.

Greetings,

      Heiko.

arendst, 2017-08-13:
@altelch Heiko as noted above I did some rewrites to the hue code to:

- get Philips hue apps working on my phone (mostly still fails. only the non color app dimmer function seems to work but it doesn't send hue data;  it only send xy color data which we do not support)
- integrate Sonoff Led with hue

As I do not own an echo I can't test Alexa.

m3zm3riz3, 2017-08-27:
I get the same issue, Alexa is not discovering the Hue emulated devices but i can use the "old" hue app to control them.

chinds185a, 2017-09-09:
Anymore news on this? kind of makes this device/firmware pointless for what I purchased it for.

altelch, 2017-09-09:
Hi,

did a merge request to fix this. Hue emulation and detection with 
alexa/echo is working again after this fix. See Pull-Requests.

Greetings,

      Heiko (altelch).

chinds185a, 2017-09-09:
Hi, Thanks, will this work as an OTA update or will I need to re-flash the device?

altelch, 2017-09-09:
Hi,

I'd guess Theo will release a 5.7.1 soon which should be OTA updateable. 
No need to flash.

Greetings,

      Heiko.

arendst, 2017-09-09:
Working on it . Hold tight.

Courty40, 2017-09-09:
I've got several Hue emulated and wimo emulated devices controlled via Alexa so will keep an eye for 5.7.1
Will test a couple of them once it's released..

Courty

chinds185a, 2017-09-09:
Nice. Happy to test this too once’s it’s out. Thanks all.

altelch, 2017-09-09:
Fixed and testet. 5.7.1 works as expected.

Courty40, 2017-09-11:
I just uploaded 5.7.1b to a Sonoff BN-SZ LED ceiling Module using the web interface.

I can confirm it works perfectly with Alexa and doesn't cause an error in the HUE app..

Courty

arendst, 2017-09-11:
Thnx for feedback.

closed 2017-09-11 by arendst

nickdos, 2018-04-02:
Just tried using Hue app with 5.12.9 and the Hue app wanted me to update the firmware on the Sonoff, as per this thread. Hue app running on iOS with version 2.19.1 (9508). Makes the Hue emulation mode useless and also completely locks up the Hue app, as there is no way to remove the new "bridge" it wants to update. Only work around is to delete the app and reinstall again.

austinlg96, 2018-04-18:
@nickdos you'll probably want to follow issue #2431