arendst/Tasmota #699: Disable relay option

pascalsaul, 2017-08-06:
In some cases I don't want to use the internal relay because of soft switching.

RelayState | 0 / off | Disable use of relay

neographikal, 2017-08-06:
This would be really really handy! I use a touch as an activator for my garage door, just to send a mqtt command. The relay doesn't have to switch (it would be nice if that could be done through mqtt separately from the button)

arendst, 2017-08-06:
Start reading the wiki. Get to know avalable commands. Search the issues and close this one please.

davidelang, 2017-08-06:
hint buttontopic

closed 2017-08-06 by davidelang

balk77, 2017-08-11:
I was looking for the same functionality; using my POW to measure power consumption of my washing machine which I don't want to switch off via the relay :)

Issue the _PowerOnState_ command with value _4_ and it is solved.

For instance as follows:
http://192.168.0.32/cm?cmnd=PowerOnState%204

See https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

pascalsaul, 2018-07-21:
When you use HA with Autodiscovery you get a nice and extra device added automatically... With PowerOnState 4 you can't control the relay but it is always on.

[image]

I would like to request a PowerOnState | 6 | Turn relay(s) off and disable further relay control

Fusseldieb, 2019-01-15:
> I would like to request a PowerOnState | 6 | Turn relay(s) off and disable further relay control

Me too. Always on is just energy consuming. I don't know why nobody has thought to put a "off" option there too.

blark, 2022-01-06:
> > I would like to request a PowerOnState | 6 | Turn relay(s) off and disable further relay control
> 
> Me too. Always on is just energy consuming. I don't know why nobody has thought to put a "off" option there too.

This ^^ Please consider this. It is logical always off functionality if there is already always on.

ascillato, 2022-01-06:
It is possible to set relays as `relay_i` ( inverted ) and use `poweronstate 4`

ascillato, 2022-01-06:
Besides that, you can define output_low as gpios instead of relay.

blark, 2022-01-06:
> It is possible to set relays as `relay_i` ( inverted ) and use `poweronstate 4`

Good idea, thank you @ascillato. I had thought of just removing the relay from the config but wasn't sure what that would do. 

I just set the following config: 

```
PowerOnState 0
SetOption73 1
```

And the button no longer toggles the relay, power always stays off. I guess this is yet another way to achieve this, I didn't think of it earlier.

barbudor, 2022-01-06:
With this, it is still possible to change the relay by sending a POWER command
If you want to totally disable that, use "OUTPUT_LOW" instead of "RELAY"

blark, 2022-01-07:
> With this, it is still possible to change the relay by sending a POWER command
> If you want to totally disable that, use "OUTPUT_LOW" instead of "RELAY"

Thanks! In my case I'm only worried about the physical button, so this does the trick.