arendst/Tasmota #700: Option for secundair GroupTopic

pascalsaul, 2017-08-06:
I would to create a circuit so I can switch a device (e.g. a bulb) with multiple devices (e.g. multiple Touch). I know there is an option to specify a GroupTopic but that topic I allready use to manage all the Sonoffs. 

Is it possible to create a secondair topic for use with soft switching? 

Switching on/off through my domotica system works:
cmnd/touch_zolder/POWER 1
stat/touch_zolder_boven/RESULT {"POWER":"1"}
stat/touch_zolder_onder/RESULT {"POWER":"1"}
stat/touch_zolder_boven/POWER 1
stat/touch_zolder_onder/POWER 1

cmnd/touch_zolder/POWER 0
stat/touch_zolder_boven/RESULT {"POWER":"0"}
stat/touch_zolder_onder/RESULT {"POWER":"0"}
stat/touch_zolder_boven/POWER 0
stat/touch_zolder_onder/POWER 0

But I loose the synchronisation when I use it with one of the physical buttons....

If the soft switch can send the MQTT message over the Secundairy Topic we're there: 

From the physical button from the Touch I would like to have a MQTT message like:
cmnd/touch_zolder/POWER 1
stat/touch_zolder/RESULT {"POWER":"1"}

neographikal, 2017-08-06:
+1 would like this :)

arendst, 2017-08-06:
Start reading the wiki. Take a look at available commands like buttontopic.

pascalsaul, 2017-08-07:
I did use the search but couldn't find anything so that's why I created this issue. Later I saw the hint 'ButtonTopic' which helped me a bit. It is imporsonating a other switch by sending the TOGGLE command  I debugged. That did help a bit but even on Google there ain't really a lot of hits ;) Maybe an idea to explain this on the Wiki how to use this 'ButtonTopic' properly then? I can write a wiki page if you want after I fixed it myself in Pimatic :)

closed 2017-08-08 by pascalsaul