arendst/Tasmota #701: PulseTime issue

jhanitesh, 2017-08-07:
I am using a basic SonOff with Tasmota 5.3.0 to use as a momentary switch. I set up the PulseTime to 20 for 2 second. Everything works fine till the Sonoff restarts (either due to wifi, power issue or manual restart). When it restarts it turns the power on to the output (for 2 sec as set in PulseTime) even though the PowerOnState is set to zero. Tried everything from the wiki and all option and used another sonoff device and result is the same. What can be done so that when sonoff restarts it does not turn on the Power for the pulsetime?

mkh595, 2017-08-07:
I am using this Pulsetime on Wemos and after restart are relays Off.
PowerOnState = 0
PowerRetain = Off
PulseTime = 130

Do you use MQTT with retain flag?

jhanitesh, 2017-08-07:
I reinstalled the tasmota software again and found it fixed the issue only to realize that when I use gpio14 as switch1 and run "switchmode1 2" and then "switchtopic alarm", I am back to the same problem. Somehow the switchtopic is causing the problem. Even when I change the switchtopic to 0 it does not fix it. 
PowerOnState is 0
PowerRetain = Off
Pulsetime = 130.
I don't think I am using MQTT with retain flag. Is there a way to verify that ?

mkh595, 2017-08-07:
I am not using switch.
They are defined as relay1-3. 30 seconds is for security reason. Control unit sends each 20 seconds "Power ON". If there is connection problem between control unit and relays then relays are switched off after 30 seconds and remain off also after reboot.

For MQTT test you can open client on desktop or mobile and subscribe for same topic like device.
Example: I have full topic "home/%prefix%/%topic%/" therefore subscription home/# will send me at beginning all retained messages.

[image]
[image]

jhanitesh, 2017-08-07:
When subscribing to the device I get these 2 retained messages. How can i turned off this retained messages (I am using mosquitto) ? On the tasmota side all MQTT_BUTTON_RETAIN =0, MQTT_POWER_RETAIN=0, MQTT_SWITCH_RETAIN=0

$ mosquitto_sub -h localhost -t "tasmota/sonoff6/#" -v

tasmota/sonoff6/tele/LWT Online
tasmota/sonoff6/cmnd/POWER ON

davidelang, 2017-08-07:
see the wiki page on PowerOnStateConfiguration

https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration

jhanitesh, 2017-08-07:
@mkh595 you are correct this is a retained message problem. I cleaned up the retained messages from mqtt broker and it solved the problem. However now I need to find a way to put the setting where messages are not retained in the mq broker. Thanks for your help.

jhanitesh, 2017-08-08:
@davidelang I made changes to my openHAB mqtt config file. It was set to retain =true. I think this was causing the problem. Changing it to false sober the issue permanently. Thanks !!

closed 2017-08-08 by jhanitesh