arendst/Tasmota #702: reverse of PulseTime, ResetTime

dorijan, 2017-08-08:
Hi to all!
Great project, and I would like to add reverse of PulsTime, something like ResetTime, for resetting power to some bad devices. Does it already exist and I am not aware of? As far as I know, PulsTime does not work in this direction, to power off and then turn the power on?
Thank you
p.s. what would be the best way to do this? Write in sonoff.ino? or define it somewhere else?

arendst, 2017-08-08:
Best way is to tell your home automation tool to toggle power at certain time.

dorijan, 2017-08-08:
I am aware of that, but what if you want to reset router/ap? then there is no way to again turn on again the rotuer. Some AP know to slow down after some time or cause problems, so I would use this as a way to reset them. There is maybe way to access the sonoff with one packet that would trigger ResetTime, but not to power it again externally.
As I said, maybe I would write this myself, maybe even write some ping trigger, so it would power off and on if there is no ping, but where would be the best place to write this code?

arendst, 2017-08-08:
Use command Backlog:

``Backlog power off; delay 40; power on``

dorijan, 2017-08-08:
Great! You are a genius! This is just what I wanted. So now all I have to do i write a ping trigger for a command :)

closed 2017-08-09 by arendst