arendst/Tasmota #708: Send status messages via HTTP to configurable target

electrontube, 2017-08-08:
TASMOTA devices can easily be controlled via http commands.
To get the current status of a device via http, it needs to be "polled"

For more flexibility to integrate TASMOTA into existing environments (with no MQTT capabilities) and to avoid "polling" it would be great to have an option to simply send status messages (just switch status ON/OFF) to a configurable target address (IP / HostName)

I would expect the status to be transmitted at: 
power on
boot / reboot
wifi connect / reconnect
every state change

Is there probably already way I have overseen, or are there chances this could be implemented?

SteveDinn, 2018-01-11:
I'm surprised there are no responses to this. I'm looking for an answer to the same question. I have no MQTT instance, and I'm looking for a way to know the state of the switch, preferrably via HTTP.

webtop, 2018-01-11:
Where is your webserver running? Are you not able to install MQTT, there, or is it by choice?

maciejtarmas, 2018-01-11:
This might be useful for example to interface with a Vera Z-Wave controller. There is no way to install MQTT, but it can send and receive HTTP requests.

SteveDinn, 2018-01-11:
@webtop I just don't have the means to install MQTT currently, but that has nothing to do with the question.
@maciejtarmas I think you are misunderstanding me. I can issue HTTP requests just fine on my own. I want to know 2 things:
1. Can I hit a URL on the tasmota device's web interface to get the status of the relay? Preferably with a JSON or XML response, so I don't have to parse an HTML page?
2. Can tasmota be configured to ping an existing URI with its current status every time one of the events in the original post occurs?

maciejtarmas, 2018-01-11:
@SteveDinn 

The Vera case is your #2.

SteveDinn, 2018-01-11:
@maciejtarmas Ah, so the tasmota/sonoff device notifies the z-wave controller of its state changes? And then that controller can ping a configurable URL? I'd rather not have to buy a piece of hardware for this, but I'll keep it in mind.

webtop, 2018-01-11:
The MQTT output is in JSON format, so what you'll need to do is send the data over port 80 instead of 1883. In the user_config.h file, change the MQTT port to 80, and watch what comes across.

davidelang, 2018-01-11:
you will have to roll something yourself. Tasmota is designed to work with a 
MQTT server, it does have the ability to send commands to it via http, but has 
no ability to send out messages via http when something changes (although it can 
send things via syslog)

The ram and flash on the device is very limited, so this feature has been 
requested many times and will not happen (so that the primary functionality of 
this firmware, operation of as many things as possible in a mqtt environment) 
can be supported.

davidelang, 2018-01-11:
On Thu, 11 Jan 2018, SteveDinn wrote:

> 1. Can I hit a URL on the tasmota device's web interface to get the status of the relay? Preferably with a JSON or XML response, so I don't have to parse an HTML page?

yes, see the status commands

> 2. Can tasmota be configured to ping an existing URI with its current status every time one of the events in the original post occurs?

no.

davidelang, 2018-01-11:
On Thu, 11 Jan 2018, Paul Allsopp wrote:

> The MQTT output is in JSON format, so what you'll need to do is send the data over port 80 instead of 1883. In the user_config.h file, change the MQTT port to 80, and watch what comes across.

what you will get is a mqtt message that contains JSON inside of it, not JSON, 
and not a HTTP message

webtop, 2018-01-11:
Yeah, I just took another look at it:

webtop, 2018-01-11:
@SteveDinn http://YOUR-IP/cm?cmnd=status will show you what you have to work with.

SteveDinn, 2018-01-12:
Thanks @webtop. That's what I was looking for. All it mentions on the 'commands' page of the wiki is Power On, Power Off, and Power Toggle. Nothing to query current status.

davidelang, 2018-01-12:
every command on the commands page will work via http

closed 2018-05-26 by stale[bot]