arendst/Tasmota #709: Sonoff S20 - Domoticz/ou - Where is my normal Toggle on/off Function?

FabianHM, 2017-08-08:
Why my S20 changed the normaly toggle on/off function to Domoticz, what I didn't need?
[image]

Normaly I see there on/off
[image]
What happend?

arendst, 2017-08-08:
Well you could start to mention that you might have installed a new version...

FabianHM, 2017-08-08:
All my devices have the Program version | 5.3.0... an the others work fine....

arendst, 2017-08-08:
But did you upgrade this one just a minute ago?

arendst, 2017-08-08:
Still no answer to my question....

Anyway, try to backup the confuration and then restore the configuration.

FabianHM, 2017-08-08:
The answear is NO, I didn't update this device. 
Now I have tried to backup an configuration from another device and plan to restore it on this problem device. It doesn't work, I get this message:
[image]

arendst, 2017-08-08:
Doesn't work with firefox. 

Try chrome and keep using the same device. So backup config of broken device and restore config just made from broken device.

FabianHM, 2017-08-08:
It works with this restore way perfekt as before, perfect!

I like your Project very much and your fast support is great, thanks for this great Tool Theo :-)

closed 2017-08-08 by FabianHM