arendst/Tasmota #710: Got it to work, now I want to change mqtt topic but I can't re-flash

tr1plus, 2017-08-08:
Hello

Not sure if I missed anything in the documentation or in another issue here, but I can't seem to figure out the following problem.

I flashed my sonoff initially with as little settings as possible. I also disabled the webserver since I'm using MQTT over TLS.
Now I want to change my topic (and in about 1-2 months I will need to change wifi parameters)n but when I make the change in the config file I get the general errors that arduino IDE can't flash to the device.

web interface and OTA are also not an option, since they are not active/configured for me.

Sonoff is in flash mode as the green LED doesn't blink. If I don't press the gpio0 button sonoff starts as usual.

Any clue what I'm missing. Am I forced to use the tool to clear my whole flash memory? I would like to do as little actions as possible since I already bricked my first sonoff a while back and I am happy I got this one to work.

Thanks!
Joren

arendst, 2017-08-08:
Why don't you change topic via mqtt?

tr1plus, 2017-08-08:
I tried serial but couldn't get it to work. However it would be nice to know a way to re-flash in case i need to change multiple settings.

Can I also change wifi settings, device name and dhcp/static ip settings via mqtt/serial?

Thanks
Joren

arendst, 2017-08-08:
Yep.

Serial should be easy. As long as it is not a Pow just connect Gnd, Rx and Tx to your pc, start arduino ide, select the correct serial port, open the serial console, set speed to 115200, send cr and lf and you're good to go.

Press enter and your sonoff should respond. If not, enter command seriallog 3

tr1plus, 2017-08-09:
Hey. I will take a look at that this evening (at work now for the next ~9hours)
For connecting to serial, you only describe tx, rx and ground. Is connecting the power not required. I think it won't boot in that case. I suppose I should just connect the complete FTDI adapter?

But to get back to my original question. Is there a way to overwrite this firmware with either a new upload or a complete different firmware? (this one seems fine, but just for my reference) And if so, how, since I can't get it to work the "normal" way.

Thanks
Joren

arendst, 2017-08-09:
Well as "anyone" else is able to get it to work in the "normal" way that's the only way supported. "Normal" way is thru Arduino IDE, PlatformIO upload or Espressifs esptool (see Wiki > Troubleshooting how to use the esptool).

You can always overwrite the firmware with a new or completely different firmware as it is hardware controlled and has no relation to the current firmware.

Regarding the serial interface, as long as your device is not a Sonoff Pow (which connects Gnd to AC life wire) you can safely connect the device to AC and have serial connection made with the FTDI using only the Gnd, Rx and Tx signals. This way you have total control over your device and can change almost any setting using the available commands.

tr1plus, 2017-08-09:
Hey, Thanks, I'll try that tonight. Something must have been connected badly if I couldn't flash then.

I'll keep you posted.

Thanks for the great software. You did an awesome job!

Kr
Joren

tr1plus, 2017-08-09:
I'm so confused. Re connected everything today and was able to flash right away. No clue what happened yesterday.

Thank you for your time.

KR
Joren

closed 2017-08-09 by tr1plus