arendst/Tasmota #712: Why is a connected serial programmer dangerous, but a sensor is not?

xSunshin3x, 2017-08-08:
hey guys,

not sure if the issue section is the right place for my question - since this is not really an issue:

i read the [Hardware Preparation](https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation) where it says: 

> Do not connect AC power and the serial connection at the same time

which makes perfectly sense for me,

**but**

why isn't it dangerous (from the electrical view) to connect an external sensor, which obviously shares the same ground and vcc pins on the board as the serial connection?

thanks in advance and regards,

Sun

pascalsaul, 2017-08-08:
I guess this is to prevent that you connect two power supplies at the same time ;)

You can power your device through a FTDI cable with 3.3V and get a serial console and flash your device.

When you connect also your mains you have two supplies at the same connected; one from the mains and one from the FTDI

The mains is also providing the 3.3V through a internal circuit. So when you connect a sensor it get the power supply through the mains which convert it to 3.3V and supplies the sensor with this. 

You could power it with mains and connect a serial console but only use the RX and TX, not the GND and VCC.

xSunshin3x, 2017-08-08:
aaah. So it basically just causes an overvoltage on the FTDI components since the 3.3V provided by the mains are heading directly to the connected FTDI board, right?

didn't saw the forest for the trees :D guess it's time for bed.

davidelang, 2017-08-09:
the problem is that the sonoff POW doesn't power the electronics relative to 
ground, but relative to line voltage. So if you connect to both FTDI and line 
voltage, you get line voltage applied to your laptop's ground, which tends to 
fry things.

kruebsam, 2017-08-09:
AFAIK the same (line voltage at logic) applies for the normal SONOFF and of course the SONOFF TH. All of them have ciruits that convert the 110...230V down to 5V (for the relais) and 3.3V for the remaining ciruitry.

arendst, 2017-08-09:
@kruebsam they convert AC to DC but they do not connect one of the AC wires to the DC ground wire as most power supplies do.

The Pow does connect AC to Gnd in order for the current sensor to work.

kruebsam, 2017-08-09:
@arendst thanks a lot for that clarification that I had not been aware of.

closed 2017-08-09 by xSunshin3x