arendst/Tasmota #713: Rf bridge

prahjister, 2017-08-09:
Do you have some directions for the rf bridge? Don't see in wiki

mihalski, 2017-08-09:
Are you referring to this?

https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#sonoff-rf

arendst, 2017-08-09:
It's function can be found on iteads website.

I will write a procedure to flash  the tiny box. Itls just a matter of opening the thing, throw the tny  switch and connect gnd, 3v3, rx and tx to the biggest header. Hold the key and apply voltage to the box.

It's now in firmware load mode.

arendst, 2017-08-09:
See https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433

prahjister, 2017-08-09:
Awesome thank you...would it make sense to reprogram the other chip to just pass on what it receives vs this learning process. The ideal way in my mind is like this project below which is great and what i am currently using.
https://github.com/1technophile/OpenMQTTGateway

For IR you can do it 1technophile's way or I did fork a while back (please dont make fun of my programming skills....they suck) 

http://www.cnx-software.com/2017/04/20/karls-home-automation-project-part-4-mqtt-bridge-updated-to-use-ys-irtm-ir-receiver-transmitter-with-nodemcu/

https://github.com/prahjister/433toMQTTto433_ESP8266

arendst, 2017-08-09:
I don't think re-programming the RF microcontroller will just pass on what it receives as between the RF microcontroller and the antenna is the Syn470R chip. YMMV.

prahjister, 2017-08-09:
I have to admit i am weak in this area but looking here
https://www.itead.cc/wiki/images/3/3e/Sonoff_RF_Bridge_433_Schematic.SCH.pdf

If the efm8 chip can be reprogrammed the 433mhz transmitter and receiver connect via rdata and tdata to the chip. 

After a little googling I think reprogramming the efm8 is the issue and not done easily

olenchenko, 2017-08-16:
Hi.
First of all thanks for your hard work!
Yesterday I have flashed my newly purchased Sonoff RF Bridge 433 (I have tried it with stock firmware and it was working fine). I must say this is not the first Sonoff device I am flashing (I have flashed couple of Sonoff Basic and couple of Sonoff POW successfully with your firmware before). I am using Arduino IDE method and yes I fave changed Flash Size from 1M (64K SPIFFS) to 1M (no SPIFFS). Compilation and upload were successful (no error) but now my Sonoff RF Bridge 433 is not starting - only the green light lids and I can not see the device connected to my Wi-Fi (I have checked wi-fi settings in user_config.h several times and I have uploaded to Sonoff RF Bridge 433 with no error several times) and I am checking on my router (I work as a sysadmin so I do have some networking knowledge).
I am connecting to 4 serial pins marked red (J2):
[image]
though there are other 4 pins marked 3V3, Rx, Tx, GND on the connector (J3) on top of RF switch.

Any suggestions?
Thanks

olehs, 2017-08-16:
I had the same problem.
You have to reset to default setting after you changed the user_config.h
Hold the button for 6 seconds

arendst, 2017-08-16:
@olenchenko and in addition to your IDE settings you did use Flash Mode Dout as explained in the wiki.

olenchenko, 2017-08-16:
@olehs Thanks for fast reply.
If you mean pressing the button for more than 4 seconds when powering on the device after firmware upload - I have tried that with no success (I still do not see any DHCP requests from Sonoff on my router though I have set my Wi-Fi SSID and passphrase in user_config.h).

@arendst Good point. I have probably used DIO - I will check when I get back home (don't have my laptop with me right now). Thanks for your help.

olehs, 2017-08-16:
@olenchenko Actually you have to hold the button for over 4 seconds after Bridge has booted. See https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

olenchenko, 2017-08-22:
Hi.
Sorry for delay. @arendst - I was using DIO Flash Mode, I have changed it to DOUT and everything works fine now. My mistake - I have missed that in readme.

Thanks again and keep up great work!

closed 2017-08-22 by arendst