arendst/Tasmota #714: Is it possible to access the analogue pin on the Wemos D1?

mihalski, 2017-08-09:
I wanted to be able to read the value of a variable resistor which requires the analogue pin but I can't even find the analogue pin bin the device configuration.

Am I looking in the wrong place or is this just not a feature available in Tasmota?

arendst, 2017-08-09:
The problem with the analog pin is that changing it's function can only be done at compile time. Tasmota default function is reading power supply voltage.

In your case you will need to disable it in user_config.ino and recompile the code.

The analog value will then pop up as a sensor value.

mihalski, 2017-08-09:
Thank you.
Maybe a mention on the wiki under connecting sensors could mention this from the user_config.ino

```
#define USE_ADC_VCC   // Display Vcc in Power status. Disable for use as Analog input on selected devices
```
I totally missed it.

closed 2017-08-09 by mihalski

martinjuhasz, 2018-01-30:
Can someone guide me where this A0 pin acutally is on the sonoff basic? would like to add an analog photo resistor.