arendst/Tasmota #716: Hacking a Sonoff basic : external LED and button

PeggyFree, 2017-08-09:
Hello,

I want to hack my basic sonoff in order to link it to **two** push buttons of this type : 

https://aliexpress.com/item/Angel-Eye-Metal-LED-illuminated-Momentary-16mm-Push-Button-Switch-Car-Dash-12V-red-blue-green/32697109472.html

**Push button wiring** : I can hack the existing push button or use the GPIO14, using the resistor hack to avoid noise. Please tell me which is best between GPIO14 or the existing button. The two buttons will be in parallel on the same input (using separate inputs would be an extra feature, for now they are supposed to trigger the same event). 

**LED wiring** : I'm thinking about hacking the existing sonoff LED. What voltage is it? As you can see on the above link, I can chose between a 3V, 5V or 12V LED button. I'll order as soon as the voltage is known.

Thanks you for your help

[image]

arendst, 2017-08-09:
Take a look at the schematics of the Sonoff Basic as available in the Wiki from iTead to find out the voltage of the led.

PeggyFree, 2017-08-09:
Thank you for you help Thomas. My electronic lessons are long gone. I see that the LED is linked to a mosfet with a 3.3V on the drain. I don't know wether the two 1k resistors (R2 and R3) are interfering on the resulting voltage given to the LED.

Link to the schematics if anybody is interested : https://www.itead.cc/wiki/images/f/ff/Sonoff-Schematic.pdf

mkh595, 2017-08-09:
I have this button in 3V version. It has already inside resistor to limit current to 20mA at 3.0 V. If you only change original led with button's LED it will not light enough because of resistor R3 in Sonoff.
Version 5V or 12V will not light because built-in resistor is too high for 3.3 V in Sonoff.

PeggyFree, 2017-08-09:
All right, so how can I do ? Should I hack the 3v button and remove the internal resistor? If I do so, may I plug two led buttons in parallel ?

PeggyFree, 2017-08-09:
Another option is getting the 220 version and plug it directly on the output, but I don't  want to play with main current. And I still expect to make it bright when the Sonoff is off.

mkh595, 2017-08-10:
You cannot remove resistor from button because it is in unassembled plastic. You can remove resistor R3 from Sonoff (it is so small) but you have to be sure that built-in transistor is cappable to deliver 20mA.

closed 2017-08-11 by PeggyFree

TommySharpNZ, 2017-09-08:
Did you get any further with this? It's exactly what I'm trying to do :-)

PeggyFree, 2017-09-08:
No, still waiting for the switches and other priorities.

PeggyFree, 2017-09-14:
It's working. I simply used +3.3v , gnd, gpio14

TommySharpNZ, 2017-09-14:
@PeggyFree Perfect! Any chance you could do a write up somewhere?

PeggyFree, 2017-09-14:
I'll try to do this. But I did à direct wiring between 3.3v and Led, and gpio14 -> push Button ->  gnd

http://reho.st/preview/self/2977b1a89608aed1cb3a508a8654f065f97ebee9.jpg

TommySharpNZ, 2017-09-14:
So does the LED light up when your lights are on or off?

PeggyFree, 2017-09-14:
Always on.
I wanted it ON when the relay is OFF, but for now I am satisfied with this first release.

xeroiv, 2017-12-29:
@PeggyFree did you check to see how much current draw the LED has in your setup?  I was hoping to do something similar and perhaps use the pwm GPIO pins to control the brightness, but didn't know if the GPIO could source/sink enough current for this setup.

Also could you provide some more details about the exact wiring? pictures, schematic, part numbers, etc to perhaps put in the wiki?  I'm sure many people would like a tutorial on what to buy and how to wire it up (myself included).

PeggyFree, 2017-12-29:
Hello, sorry but it will be a short answer from my phone. I did a direct
wiring of the gpio14 + gnd on two buttons bought here :

http://s.aliexpress.com/YNRrYZb2

I bought the 3.3v version in "color" rainbow. The color fading is too fast
so I do not recommend this "color".

I  wired the +3.3v + gnd for the button's led. I didn't hack the button nor
the sonoff.

akpuggy, 2018-01-05:
Hi @PeggyFree . Can you please clarify a little which pins you used on the LED button since there are 5 to chose from? I want to do the same, except using the original gpio0 button since I want to do this on a device with original firmware. Thanks!

PeggyFree, 2018-01-05:
I don't remember, my cables were not soldered on the led button and they
went off last month. I'm sorry I don't have time for you, but honnestly it
is very very basic. I have nothing  to share, the button wiring will be
found easily by testing. You won't fry anything.

ramgopal5, 2018-01-27:
I’m trying to do the same and I like the push button config akpuggy in first attempt. Wanted to confirm if we do that and connect the Onboard LED to the push button LED Is there enough tocreate an inverted effect. Meaning the onboard led works as usual BUT the push button is off and when the switch is off the push button LED is lit ??

akurej, 2018-02-05:
Hi akpugg. In your 1st attempt, can you please let me know the LED state on the push button?
Does it light up when the switch is on and light off when the switch is off?

Any idea if it's possible to mirror the on board LED with push button LED and at the same time the push button works as the on board switch?

sirs2k, 2018-05-24:
Any of you seen this video?
https://www.youtube.com/watch?v=qKtIMo_s520&t=128s

He's wired the LED to the back of the board and apparently it's 5V.

akpuggy, 2018-06-23:
I think I finally figured it out. Posted schematics here: https://community.home-assistant.io/t/light-switch-with-led-push-button-and-a-pir-sensor-on-sonoff-basic/57669