arendst/Tasmota #718: Device responding only once to a status query

CrazyIvan359, 2017-08-09:
I am setting up a rule in openHAB to trigger status messages from all Sonoff devices in my home when the system restarts. The rule works fine, but I noticed that the devices will only respond to the first blank message in the 'POWER' or any other topic requesting status information. I have to send a power on or off command and then the device will respond to a blank message on the 'POWER' topic again, but still only once. I have tested this using openHAB and with mqtt-spy to send the blank messages with the same results. Once the device seemed to lock up and would not respond to any messages at all until I sent it an ON command via http, then all of the status requests I had sent were replied to all at once.

arendst, 2017-08-10:
Cannot reproduce. 

My mqtt-spy results are always as expected: ``cmnd/pow2/power`` returns ``ON`` repeatedly.

Enable (serial) logging and see what happens (``seriallog 3`` or ``weblog 3``).

CrazyIvan359, 2017-08-10:
I will turn logging on tonight and see if I can reproduce it. I just tested it again and it is working normally with repeated blank messages from openHAB and mqtt-spy.

I pulled some charts from mqtt-spy to show what it was doing yesterday

my messing around with it over about an hour:
[image]
zoomed in (busy section on the left side of the above chart)
[image]

CrazyIvan359, 2017-08-20:
Well I haven't seen anything in the logs and have not been able to reproduce this issue since then, so closing this issue.

Thanks for you quick reply

closed 2017-08-20 by CrazyIvan359