arendst/Tasmota #720: sonoff does not connect after light off

moonfarms, 2017-08-10:
I am living in a place in Africa where there is often light off (erratical electricity suply). I have noticed that when power (electricity) is back and everything rebooted, due to the delay of the wifi router to boot the sonoff module will not connect to wifi unless, after wifi router is back online, I power off/on the power of the sonoff module.
I understand it is not a general issue but related to different environmental? condition.
Thanks in advance

arendst, 2017-08-10:
What is the value of wificonfig? Try to set it to 4 to retry wifi connection.

moonfarms, 2017-08-10:
Thank you for your prompt help.
2morrow i will look where to config wificonfig and get back.
have a nice morning/day/evening wherever you are.

closed 2018-05-07 by stale[bot]