arendst/Tasmota #721: How to use Sonoff Dual for Rolling Shutters ?

zibous, 2017-08-11:
First of all, thank you for the great firmware. I flashed my sonoff dual 
and am very pleased with how everything works.

I try to set the SwitchMode (shutter1up/shutter1down, shutter2up/shutter2down)
but i do not work. It very importent that never bouth relais will 
be switched on at same time.

12:48:44 CMND: SwitchMode=2
12:48:44 MQTT: stat/SD001/RESULT = {"Command":"Unknown"}

Also the Web Command to nothing ?
http://sonoff/cm?cmnd=Power%20On
http://sonoff/cm?cmnd=Power1%20On

Any advice for the GPIOs for the Dual to add a DHT22 Sensor ?

PeggyFree, 2017-08-11:
You shloud look at this : https://github.com/arendst/Sonoff-Tasmota/pull/519

zibous, 2017-08-11:
Thanks, but i can't find a working information. When i use the "Module":"Shutter" i can't control the Switch with MQTT or with web command:
14:14:45 MQTT: tele/SD001/INFO1 = {"Module":"Shutter", "Version":"5.5.2b", "FallbackTopic":"DVES_1E1995", "GroupTopic":"sonoffs"}
14:14:45 MQTT: tele/SD001/INFO2 = {"WebserverMode":"Admin", "Hostname":"SD001-6549", "IPaddress":"10.1.1.243"}
14:14:45 MQTT: tele/SD001/INFO3 = {"Started":"Software/System restart"}
14:14:53 MQTT: tele/SD001/STATE = {"Time":"2017-08-11T14:14:53", "Uptime":0, "Vcc":3.156, "Wifi":{"AP":1, "SSID":"McWille", "RSSI":92, "APMac":"C4:B3:01:DC:55:B6"}}
14:15:53 MQTT: tele/SD001/STATE = {"Time":"2017-08-11T14:15:53", "Uptime":0, "Vcc":3.106, "Wifi":{"AP":1, "SSID":"McWille", "RSSI":100, "APMac":"C4:B3:01:DC:55:B6"}}

arendst, 2017-08-11:
Module Shutter?

zibous, 2017-08-11:
Hi,

I try this, but it do not work.

{ "Shutter",         // Sonoff Dual as Shutter Switch (ESP8266)
     0,
     GPIO_TXD,         // GPIO01 Relay control
     0,
     GPIO_RXD,         // GPIO03 Relay control
     GPIO_USER,        // GPIO04 Optional sensor
     0,
     0, 0, 0, 0, 0, 0, // Flash connection
     0,
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  }
[image]

pascalsaul, 2017-08-11:
I'm also looking for a shutter like function to replace my thermostat. Only I need 3 states: 1 idle (no relay), 2 heating (relay1), cooling (relay2). In idle state you can turn on relay1 or relay2. If relay1 is on, you can't turn on relay2. You always need to go in 'idle' state first or you should be able to switch between heating/cooling. My dual is on his way so I didn't play with it yet.

ionciubotaru, 2017-08-12:
Try:

`backlog power1 OFF;power2 ON`

In this way you will never get both relays ON

zibous, 2017-08-12:
Thanks for the Information from ionciubotaru, 
works in Console and with homeassistant MQTT.

``
## -----------------------------------------
## homeassistant MQTT cover with SONOFF DUAL
## -----------------------------------------
cover:
  - platform: mqtt
    state_topic: "tele/COK001"
    command_topic: "cmnd/COK001/Backlog"
    name: "Rollladen"
    qos: 0
    retain: true
    payload_open: " power1 OFF;power2 ON"
    payload_close: "power1 ON;power2 OFF"
    payload_stop: "power1 OFF;power2 OFF"
    state_open: "OFF"
    state_closed: "STATE"
    optimistic: false
``
	
Works Cover UP/DOWN/STOP works with MQTT
	
Problems:
How to get the State ?
Can'nt use the Shutter Motor Switch (Button 0, Button 1) on SONOFF DUAL.
How can i add the temperature and humidity sensor DHT22 ??

Used:
Program version	5.5.2b
Build Date & Time	2017-08-11T15:12:49
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Sonoff Dual Module
[image]

mkh595, 2017-08-13:
`Try:backlog power1 OFF;power2 ON

In this way you will never get both relays ON`

only as long as you don't press button on device or in Web gui or don't send MGTT command for one relay

zibous, 2017-08-13:
Hi,
Thanks, 

 "backlog power1 OFF;power2 ON In this way you will never get both relays ON"  works perfect.

`command_topic: "cmnd/COK001/Backlog"
  payload_open: " power1 OFF;power2 ON"
  payload_close: "power1 ON;power2 OFF"
  payload_stop: "power1 OFF;power2 OFF"
`
Only the hardware based switch i can't use (see Switch Shutter Motor).
Also i can't find the GIPO Pin to connect the DHT Sensor.

ionciubotaru, 2017-08-13:
In the web interface select CONFIGURATION then CONFIGURE MODULE
Select DHT sensor on the DESIRED GPIO pin
If the pin is not listed you must configure it in `sonoff_template.h`

zibous, 2017-08-13:
@ionciubotaru 

Thanks for this information, but i can't find the DESIRED GPIO pin on the print layout
of the Sonoff DUAL. 

I would be glad if I could get the information, where I can connect the DHT sensor to the SONOFF Dual. Only then can I adjust the sonoff_template.h to get the sensor values.

ionciubotaru, 2017-08-13:
Read [https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual)

Use RX pin for DHT
Disable serial log before

arendst, 2017-08-13:
Forget using Rx or Tx on a Sonoff Dual!

Serial interface is being used by relay control. That's also a good reason not to use a Dual for interlocked motor control as you'll someday loose communication between the esp8266 and the relay controller and both relays will be on.

I suggest to use something different for interlocked motor control.

zibous, 2017-08-13:
@arendst,

Many thanks for the information, but does this mean that I CAN NOT USE the SONOFF DUAL for a roller shutter?

**Or just no DHT sensor ??? **

I also suspect the Sonoff T1: 1-3 Gang WiFi & RF 86 Type UK Smart Wall Touch Light Switch is not suitable for this? I would like a solution better than SONOFF DUAL. But also this product has in no interlocked mode.

[image]

arendst, 2017-08-18:
I will release global interlock starting with version 5.5.2k

zibous, 2017-08-18:
@arendst

Thanks, is working perfect.
Could I use it also for the Sonoff T1: 1-3 ?

arendst, 2017-08-18:
Yep.

zibous, 2017-08-18:
Perfect, so i can order this...

my Configuration for HA is:
```
cover:
  - platform: mqtt
    state_topic: "tele/COK001/STATE"
    command_topic: "cmnd/COK001/Backlog"
    name: "Rollladen"
    qos: 0
    retain: true
    payload_open: "power1 ON"
    payload_close: "power2 ON"
    payload_stop: "power1 OFF;power2 OFF"
    state_open: "ON"
    state_closed: "OFF"
    optimistic: false
    value_template: >-
      {% if value_json.POWER1 == 'OFF' and value_json.POWER2 == 'OFF' %} 50 
      {% elif value_json.POWER1 == 'OFF' %} 100 
      {% elif value_json.POWER2 == 'ON' %} 0 
      {% endif %}
```

Only the state is not working ??
If I change the status (up / down), then this is not immediately reported back - only at next MQTT request.

giig982, 2017-08-20:
Hi, I'm trying interlock mode with v5.6.1 and I can set Option14 via web interface. 
My problem is that I did not found an option to mantain this option after a reboot, so I have to set again Option14 every time I switch the Sonoff (Dual in my case) off/on.

@arendst is there any future release that will integrate this option in the configuration, in memory?

ionciubotaru, 2017-08-21:
After change Option14 send a `saveData` command, or set `SaveData 1` for autosave changes

giig982, 2017-08-21:
Thank you for your reply. I'll try it!

electrontube, 2017-08-21:
@arendst 

interlock works great, thanks for the new release.
What we need in addition now is some kind of calibration/position data for use with shutters.

Assume It takes 60secs for the shutter to run from fully open (100%) to fully closed (0%)
If one could enter this "max_runtime" and there was an additional topic i.e. "Position" the shutter could be send directly to a position (n%) by switching the required relay (up/down dependent on the current position of the shutter) "on" for the required period of time to reach that position.

I know, this method is not 100% precise, as "down" will most likely run a bit faster than "up", due to different load, but it should be much better than triggering relays from an external source as there are random latencies in the communication path.
In addition, only the Sonoff Module itself would "know" the current Position at an acceptable accuracy as it controls the relays.

Auto re-calibration is simple: 
when a relay is switched on for more than the calculated required time to reach the end position in the direction, the "current position" memory is set to 0 or 100 according the shutter position.

What do you think about it?

giig982, 2017-08-21:
Fyi, neither pulsetime time will be retained after shutdown and restart

davidelang, 2017-08-22:
pulsetime lets you say how long an output should remain on before it turns off.

giig982, 2017-08-22:
I know it @davidlang, but if I set a pulse time or interlock mode I'd like to retain those settings even after Sonoff restart, without the need to set them again. As I said, I tried to send 'savedata' or also 'savedata 1' to save settings in memory, without any results.

davidelang, 2017-08-23:
If the pulsetime values are lost on reboot, please file a bug to fix that rather 
than asking for a different way to do the same thing.

giig982, 2017-08-23:
I never asked a different way to do the same thing. I just added some informations to this issue that can be correlated.

Anyway, I opened bug issue #776

Fragmike, 2018-02-18:
Hi All,
finally flashed my Sonoff T1 R2 Uk Touch :-)
I´m on 5.12Sonoff-Tasmota 5.12.0b 

Sending the comand 

> backlog power1 OFF;power2 ON

in Console does not enable intelock mode
Can someone help me?

renne, 2018-03-11:
@electrontube 
@arendst 

Shutters change their radius while winding which leads to a exponential time-distance correlation. I suggest the following procedure:
1. Wind up shutter completely
2. Measure the time for closing the shutter completely
3. Measure the time for opening the shutter completely
4. Measure the time for closing the shutter halfway through

Step 4 can be used to approximate variables for a exponential function.

opendore, 2018-03-21:
I need some clarification, I installed the latest version of Tasmota 5.12.0 and I would like to command my shutters with a dual sonoff R2.

The first question I ask is:

Can I control the dual R2 sonoff with two external buttons?

I have configured the sonoff as Motor C / AC Module but I see that in the settings I can not assign anything to the buttons 1 and 2 .... Is it correct?

if I send the following command from console to test and see if relay 1 automatically deactivates relay 2 when it does not work:

22:33:06 CMD: power2 on
22:33:06 RSL: stat / sonoff / RESULT = {"Command": "Unknown"}

only the power1 on command is working.

Sorry for the stupid questions but they are at the beginning ....

Thank you all for the support!

KristerBC, 2018-03-24:
Hi everyone,
In addition to opendore's question, I would like to know if the new firmware also affects the physical buttons on the switch ?

Ex.
If relay1 is on and I touch relay2 manually... with my fingers on the wall switch, will it then automatically turn relay1 of and relay2 on?

Just want to make sure, before flashing my precious Sonoff T1 :)

KristerBC, 2018-03-25:
To respond to my own question,

The firmware also work when you physically push the buttons. I just flashed my T1 and I know have an interlocking mode which make it impossible to have both buttons "on" in the same time.

rflx18, 2018-05-01:
sorry guys, i dont follow the interlock mode, i cant find it anywhere, and can someone please past the HA configuration so only one button is pressed at one time.
thanks in advance

KristerBC, 2018-05-01:
Yeah, to be more precise, you can use the following command in the console:
SetOption14 1
It will activate the interlock mode

Make sure the that the device does not reset this option on reboot. Shutter motors don't like to have both directions activated at the same time.

Page for commands: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

opendore, 2018-05-03:
Thank you so much!!!!

It's been a long time since I could not go on with the project of my
automated blinds.

I tried the command you said and it works great.
Now the last question ..... how do I fix this option so that if you
had to restart my sonoff r2 do not lose this setting?

Thanks in advance for your support!

KristerBC, 2018-05-03:
Great that this thread could help you out.

In general, I think it should save the option automatically even after a restart and even a firmware upgrade via the web interface. For me, all my 5 T1s have kept the option on, despite manual firmware updates and without power during multiple hours(mine's are integrated in the walls).

I just wanted to point it out, because during the flash of the first device, I had it powered using my programmer through USB and it did loose the setting sometimes.
I guess it was as a result the lower voltage, but it would be great if somebody with more knowledge than me could give an opinion about this?

opendore, 2018-05-07:
How can I proceed now that I have managed to make the sonoff fuse switches work and manage via ip from my phone using the web interface.
Do I have to install a MQTT server like Mosquitto on my raspberry?
There are applications on IOS to manage more Sonoff with Tasmota that go to connect the MQTT server of the Raspberry? Has anyone done something like this? 
Can you give me some suggestions to manage the shutters from the phone?
Thank you all for your help.

GeeBee93, 2018-05-14:
@KristerBC  Are you using the T1 to control your shutters? I am currently planning to change my shutter controls at home to sonoff switches, and the price difference between the "normal" dual (http://sonoff.itead.cc/en/products/sonoff/sonoff-dual) Version and the Touch Dual Version (http://sonoff.itead.cc/en/products/residential/sonoff-t1-eu) is not that much. This would add the nice capability to control the shutters through web interface or a "manual" switch. Thanks!

KristerBC, 2018-05-15:
@GeeBee93 Yes! I am using the T1 dual touch switches to control my shutters and they work perfectly with the interlock option(Option14) activated.
I did think about using the T1 triple touch and using the middle button as a "stop"-button, but I decided, that it would be too complicated for the user.

My thought was exactly the same. I did not want the shutters to be unusable, if my Google Home or router stopped working. Also, if you have guests, they should not have difficulties knowing how to open/close the shutters.

titomontes, 2018-05-19:
Hello everyone. I just found this very useful post. 
Has anyone used the Sonoff Dual for blinds? How would the wiring go? Can an external wall switch be used?
Also, how is the T1 wired anyway? 

You´ve taken my hopes up again with this great firmware!

Thanks a lot!

@opendore: take a look at [Home Assistant](https://www.home-assistant.io/hassio/), you cam implement it in a RPi 3, and control all of your domotics from there.

closed 2018-07-18 by stale[bot]

opendore, 2019-01-27:
Yes is correct!

wowtah, 2019-08-13:
Great stuff! Can somebody post their config (also maybe the Homeassistant automations) so I can create my own? Many thanks!

Swabeats, 2020-01-28:
Any updates on this one? Available in Homekit?

ascillato, 2020-01-28:
@Swabeats This issue is very old. Please, check the docs (tasmota.com) because now Tasmota supports rolling shutters. Thanks.