arendst/Tasmota #722: How do I specify a switchtopic in user_config.h?

shrug1, 2017-08-11:
I'm working to add external switches to my project.  If I specify a switchtopic via the web UI console (_'switchtopic xxxxxx'_), everything works as expected.  But I haven't figured out how to specify a switchtopic at build time instead of at run time.  Neither

#define SWITCHTOPIC, nor
#define SWITCH_TOPIC

seem to work, and I haven't found an obvious place in the other code modules that would pick up a #define that I'd make anyway.  Can someone please help me find what I'm overlooking?

arendst, 2017-08-11:
Not supported. Use switchtopic command as you already did.

shrug1, 2017-08-11:
Well, then, I'll stop looking!  :)  Thanks much.

I'll also take this opportunity to thank you for the incredible job you've done on this project; I'm regularly amazed at how capable it is!

closed 2017-08-11 by shrug1