arendst/Tasmota #723: Mosquitto does not responds like expected

Dappie, 2017-08-11:
I have a problem that i don't understand.
If I send a command trough Moquitto like:

**mosquitto_pub -d  -t "cmnd/sonoff-5923/power1" -m on**

the sonoff does not respond.
After looking at the response from Mosquitto I notice this:
Received PINGREQ from DVES_3C1723

So I used the command:

**mosquitto_pub -d  -t "cmnd/DVES_3C1723/power1" -m on**

And that works. 
The hostname of the sonoff = sonoff-5923 (thats 1723Hex)

Am I doing something wrong?

Regards
Steef

Sonoff info:
[image]

davidelang, 2017-08-11:
the hostname doesn't matter, what matters is the topic that the sonoff is 
configured to respond to.

hostname is strictly a DNS thing, mapping a name to an IP. Once something 
connects to mosquitto, what matters is what that thing subscribes to and/or 
publishes as.

jpenninkhof, 2017-08-11:
According to your config screen your device's topic is still sonoff. Try:

mosquitto_pub -d -t "cmnd/sonoff/power1" -m on

Dappie, 2017-08-11:
Thanks!  so i need to change the name 'sonoff' in a alternative name?  Now all divices respond on 'sonoff' :-)
and 'sonoffs' in a groupname..  Is that correct?  and where in the source? (user_config.h)
These fields:
#define MQTT_TOPIC             PROJECT           // [Topic] (unique) MQTT device topic
#define MQTT_GRPTOPIC          "sonoffs"         // [GroupTopic] MQTT Group topic
#define MQTT_CLIENT_ID         "DVES_%06X"       // [MqttClient] Also fall back topic using Chip Id = last 6 characters of MAC address

jpenninkhof, 2017-08-11:
That's right. It is probably easiest to change it in the web interface instead of building firmware for each individual device.

Dappie, 2017-08-11:
Great!  next step openhab :+1:

closed 2017-08-11 by Dappie