arendst/Tasmota #724: Planned support for S30

vcazan, 2017-08-11:
Want to buy the S30 unit but see it is not supported. Any plans on supporting this device?

arendst, 2017-08-12:
Not yet but if you can provide information regarding internal connections to button and relay it should be easy to do.

It's currently only available in US version (110V).

JonSilver, 2017-08-18:
I believe these are Itead S30 UK model:
https://www.amazon.co.uk/MEAMOR-Wireless-Function-Required-Anywhere/dp/B0743BLH3G/

bobmoran00, 2017-09-30:
I have opened one up, S30 US version.  Glued case, 8285,  four pin holes labled vcc D+ D- Grd.  were no help.
Used the four holes at J1  J1=vcc, next=Tx, next=rx next(boxed)=grd.  I was able to flash blink with these as a test.  Still testing.
The unit is similar to the Amazon refrence but no USB port but that is probable what Vcc,D+D-&grd are for.

bobmoran00, 2017-10-01:
Turns out the USB is a Charging only port D+ & D- go nowhere.

closed 2018-05-07 by stale[bot]

mrubli, 2021-01-31:
For reference below are some photos of a destructive S30 teardown. As bobmoran00 pointed out J1 exposes the 8285's `EXT_RSTB`, `U0TXD`, `U0RXD` pins and ground. The unpopulated USB port is simply for power since `D+` and `D-` are tied together.

[image]
[image]
[image]