arendst/Tasmota #725: Binary Sensor Help Please

Odianosen25, 2017-08-11:
Hello,

Many thanks for your firmware, and it must be great knowing your code helps thousands of ppl like myself to achieve awesome things.

Please a quick question, how can I connect a binary sensor to the sonoff basic. Like PIR or Window/door sensors that are either closed or opened so the status can be sent to an MQTT server?

Regards

davidelang, 2017-08-12:
you configure the GPIO you connect the sensor to as a switch, and then configure 
switchmode and/or switchtopic to give you the results you are looking for.

To wire the sensor, you need to see if they apply power or just close a relay 
(there are some of each type), if they apply power, you need to make sure it's 
not above 3.3v (and setup a resistor bridge if it is)

If it's a pure relay, you wire it like you would any other switch.

Odianosen25, 2017-08-12:
Hello @davidelang ,

Thanks for the response. I do understand I have to define my GPIO14 sensor as a switch, also understand I have to set it to switchmode=2 so that it can send a message to a defined mqtt topic set by myself. I also understand I have to set the switchtopic to what I want to use.

But how to do all these is quite confusing I am afraid. I understand I can send command via http requests, web or mqtt, but I don't understand how the syntax come together. Please I just need a clearer way to achieve this, as this is quite new to me.

Kind regards

Odianosen25, 2017-08-12:
Don't worry @davidelang  thanks and I have gotten it to work now. Many thanks again and I am most grateful

mkh595, 2017-08-12:
Read in Wiki [Commands topic](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)
E.g. command Status 0 returns all information

CommodoreWhite, 2017-08-13:
Consider setting up a grouptopic and assigning all (or both in your case) your Sonoffs to it. Send the "status 0" to the grouptopic and all (or in your case, both) devices will respond with their status. Neat!

Peter

PeggyFree, 2017-08-13:
That's a great tip, I was looking for a simple way to retrieve the Sonoff
status when my server reboots

closed 2018-05-07 by stale[bot]