arendst/Tasmota #726: GPIO14 as door opening sensor

witek1308, 2017-08-12:
Hi,
I can't find this anywhere on the internet whether Tasmota on Sonoff basic would support the a door opening sensor on GPIO14, but **without impacting** the state of relay (use GPIO14 just for reporting the state, ideally over MQTT).
I simply in my garage have one Sonoff Basic with Tasmota (5.5.2) which controls some other thing but is so close to the garage door that wiring will be piece of cake, it would be perfect to report open/close state. Garage door is controlled by "1 channel inching 5v" module running Tasmota too but as I understand this one doesn't have any GPIOs to do the work. Thanks in advance for all hints.

witek1308, 2017-08-12:
I used wrong filters for searching, I found similar question just before mine. I'll give it a try therefore I'm closing this subject.

closed 2017-08-12 by witek1308

jtroberts1, 2017-09-01:
@witek1308 , can you tell me how you solved this issue?  can you upload the link you referenced.  I have switches on GPIO14 and 5 with relays on 4 and 12.  every time I remove power from the Wemos it triggers the garage door.  thanks in advance.

witek1308, 2017-09-02:
Will have a look when back at my pc. Please check issue #725 this is where I started.
When I am back I can check my config in details

witek1308, 2017-09-02:
ok the issue #438 is exactly what helped me (it's clear instruction what to do)

sabs01, 2017-09-21:
thanks is a good step in the right direction for me but cant seem to figure out how to allow home bridge to read the Mqtt message and change the status in home kit -  Using sonoff basic as a garage door opener with a reed switch. (need to be able to detect when the manual RF opened is used  rather than homekit)

witek1308, 2017-09-21:
you need to mount the opening sensor anyways (I did it). Reason: my garage door doesn't have just 2 states: opened and closed. If you press the button during opening you can stop it in between , same when closing. Therefore detecting the button was pressed will not tell you the state if this is what you need (especially if someone pressed the garage door RF or wall button  - in my case you can control the door using 3 different ways independently - sonoff, RF remote, wall button)

sabs01, 2017-09-21:
thanks for your help -- I agree this seems to be the way to go.. do you have the code you have used for this to work with homebridge and sonoff ?

witek1308, 2017-09-21:
> Data: 21 września 2017 2:34:15 PM
> Do: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> DW: witek1308 <witek_1308@poczta.onet.pl>, Mention <mention@noreply.github.com>

There is no code. I use openhab and integrated sonoff via http request

Wysłano z AquaMail dla systemu Android
http://www.aqua-mail.com

--- Wiadomość oryginalna ---

sabs01, 2017-09-21:
thanks I think I have figured it out  - have changed my config.json file to the following : 
"statusGet" :  "cmnd/alarm/POWER1" (this is the MQTT message topic from GPIO14 GND switch)

instead of 
"statusGet": "stat/sonoff/RESULT" (this is the MQTT message topic from the relay button)

this allows me to read 1 reed switch but need to figure out how to set up pins on the sonoff to read more than 1 switch, and code to read the MQTT topic for any other reed switches?? can anyone help thanks in advance.

davidelang, 2017-09-21:
you can define up to 4 GPIOs to be switches (switch1-switch4)

sabs01, 2017-09-21:
thanks David I am a noob at this can you explain how - have tried to configure sonoff as a WEMO mini and set GPIO3 as switch 2 but no joy?

davidelang, 2017-09-21:
what do you mean by 'no joy', do you mean that you can't find where to set the 
GPIO as switch2, or that it's not working as expected after you make that 
setting?

I seem to remember that GPIO3 is part of the serial port, so you need to disable 
serial logging for it to be usable as a generic I/O pin.

sabs01, 2017-09-21:
Thanks David yes I can set GPIO3 to switch2 but cant get it to work in the same way GPIO14 does - you are correct GPIO3 is a serial port- perhaps I need to use a different GPIO to make things simple what do you think? any ideas on how to access other pins on the sonoff basic?

sabs01, 2017-09-21:
Mate you are a legend I have disabled serial logging and it works as an I/O pin - reading as cmnd/alarm/POWER2 = TOGGLE
now trying to change it to read ON and OFF any ideas?

davidelang, 2017-09-22:
the sonoff basic doesn't really let you at the pins easily. If you disable the 
serial logging, you can try to use those pins as GPIOs for sensors.

otherwise, you need to solder to the pins or use a different module.

jannnfe, 2017-10-31:
@witek1308 How do you get Tasmota on your 1 channel inching 5v. Can you please have a look at my issue?
https://github.com/arendst/Sonoff-Tasmota/issues/893
Thanks!

bandric1, 2017-11-01:
Can someone please help me im getting into my hairs and its killing me:

I changed topic to another one i see it in mqtt.fx it goes to on or to off but in openhab it wont show open or closed this is my setup:

I can switch it and everything i just cant get it to show when i close gpio14 to gnd that is closed. I dont get the read out from gpio14 > openhab.

Mqtt.fx: shows this when state to gpio14 is changed:
cmnd/alarm-000B9593D/POWER2
ON

items:
Switch  T_D_Garage1 "Garage Door 1" <switch> 
[ "Switchable" ]
{ mqtt=">[broker:cmnd/WemosD1-000B9593D/POWER:command:*:default]" }

WemosD1-000B9593D topic works fine i can use this to publish cmnd no problem. I have changed the topic for switch2 via console to alarm-000B9593D for the gpio14 swich2 (10)

/* Items to represent the open/closed status of the opener */
Contact N_D_GarageDoor1  "Garage Door 1 [MAP(en.map):%s]" <garagedoor> { mqtt="<[broker:cmnd/alarm-000B9593D/POWER2:state:default]" }

In folder transform i have created en.map i put strings in there to define:
ON=closed
OFF=open
NULL=Alarm
 
Sitemap:
Switch item=T_D_Garage1 label="Garage Door 1" icon="garagedoor-closed" mappings=[ON=Open] visibility=[N_D_GarageDoor1!="OPEN"]
Switch item=T_D_Garage1 label="Garage Door 1" icon="garagedoor-open" mappings=[ON=Close] visibility=[N_D_GarageDoor1=="OPEN"]

Please if someone can help me in any way or just point me.

davidelang, 2017-11-01:
I do things like:

Switch sw01 "David's Light" { 
mqtt=">[openhab-mqtt:cmnd/sw01/POWER:command:OFF:0],>[openhab-mqtt:cmnd/sw01/POWER:command:ON:1],<[openhab-mqtt:stat/sw01/POWER:state:MAP(binary.map)],<[openhab-mqtt:stat/sw01/POWER:state:MAP(binary.map)]", 
autoupdate="false" }

this watches for changes via stat messages, and via tele messages

cat binary.map
1=ON
0=OFF
On=ON
Off=OFF
ON=ON
OFF=OFF

bandric1, 2017-11-01:
Thanks for a hint but it didnt work i tried your way. With state:MAP and no dice wont read. I just dont get it i can switch the items on/off but cant read current state dont get it i had to change topic due to unlinking the gpio14 from gpio5 so 14 dont turn 5 on when closed.

davidelang, 2017-11-01:
check the openhab logs, it should tell you where it's having trouble.

bandric1, 2017-11-01:
There is nothing in the logs i have allready checked it in openHABshare/openhab2-logs/

davidelang, 2017-11-02:
then check the mosquitto logs and see what it's showing.

Increase log detail if you need to.

you should see the stat/ or tele/ message arrive at mosquitto and what it is 
being delivered to.

Once you have openhab configured to get the correct messages, you should see the 
message arrive in openhab and what it does with it.

Paulf007, 2017-11-02:
I am running a similar setup and it took a while for me to get this figured .
Here is a Pick of the SonOff Basic 
[image]

I set the switchmode/switchtopic as per the wiki so that it works independently.
Then in Openhab.
ITEMS PAGE
Switch main_gate_gate "Main Gate" { mqtt=">[broker:nodered/main_gate/POWER2:command:*:default]", autoupdate="false" }
String main_gate_stat "Main Gate [%s]"{ mqtt="<[broker:cmnd/gate/POWER3:state:default]" }
String main_gate_stat1 "Main Gate1 [%s] " { mqtt="<[broker:tele/main_gate/SENSOR:state:JSONPATH($.Switch3)]" }
String main_gate_stat2 "Main Gate2 [%s] " { mqtt="<[broker:nodered/main_gate/SENSOR:state:default]" }
 
SITEMAP
Switch item=main_gate_gate visibility=[main_gate_stat=="OFF",main_gate_stat2=="OFF"] mappings=["ON"="Open Gate"] icon="garagedoor"
Switch item=main_gate_gate visibility=[main_gate_stat=="ON",main_gate_stat2=="ON"] mappings=["ON"="Close Gate"] icon="garagedoor"

You will see in the command line of the switch that I send the value to node red as well as recieve from node red. The reason being that I have a funting the switches POWER2 of about 500ms after the on command was receive. Also use that state update to show or hide the switch on the site map. 
Hope this helps.

bandric1, 2017-11-02:
Thanks a million "Paulf007" this one helped me finally im out off the loop. Now i see the state change but im having issue when sensor closes i get two states shown in classic-UI

this is sensor open:
[image]

this is the sensor closed
[image]

I had to change settings in my module it would not work like basic (with it GPIO rules) so i made them work like this:

[image]
 
This is my items code:
Switch main_gate_gate "MainGate" { mqtt=">[broker:cmnd/WemosD1-000B9593D/POWER1:command:*:default]", autoupdate="false" }
String main_gate_stat "MainGate [%s]" { mqtt="<[broker:cmnd/WemosD1-000B9593D/POWER1:state:default]" }
String main_gate_stat1 "MainGate [%s]" { mqtt="<[broker:tele/alarm2/POWER2:state:JSONPATH($.Switch2)]" }
String main_gate_stat2 "MainGate [%s]" { mqtt="<[broker:cmnd/alarm2/POWER2:state:default]" }

This is my sitemap code:
            Switch item=main_gate_gate visibility=[main_gate_stat=="OFF",main_gate_stat2=="OFF"] mappings=["ON"="Open Gate"] icon="garagedoor"
            Switch item=main_gate_gate visibility=[main_gate_stat=="ON",main_gate_stat2=="ON"] mappings=["ON"="Close Gate"] icon="garagedoor" 

The sitemap is like it is yours "Paul007"

This is my console point out of WemosD1-mini

00:00:17 MQTT: stat/WemosD1-000B9593D/POWER1 = ON
00:00:18 MQTT: stat/WemosD1-000B9593D/RESULT = {"POWER1":"OFF"}
00:00:18 MQTT: stat/WemosD1-000B9593D/POWER1 = OFF
00:00:22 MQTT: cmnd/alarm2/POWER2 = ON
00:00:25 MQTT: cmnd/alarm2/POWER2 = OFF

bandric1, 2017-11-02:
I made a typo while edit now it works!!!  

this is the correct sitemap:
Thanks again "Paul007"

Switch item=main_gate_gate visibility=[main_gate_stat1=="OFF",main_gate_stat2=="OFF"] mappings=["ON"="Open Gate"] icon="garagedoor"
Switch item=main_gate_gate visibility=[main_gate_stat1=="ON",main_gate_stat2=="ON"] mappings=["ON"="Close Gate"] icon="garagedoor"

If someone wants to build a GOPENER with arenst tasmota and Wemos D1-mini i can say that it works now i got some greyhairs and restless nights. But got out off it thanks to Paul007.

Upper response is full layout od settings after uploading the firmware you have to go to console:
1). Type in PULSETIME 15 (or any number your relay shield wild then be push button)
2). 2a. type: switchmode 2
2b. type: switchtopic alarm
https://github.com/arendst/Sonoff-Tasmota/issues/438

Then go and play with openhab....

Paulf007, 2017-11-03:
> Upper response is full layout od settings after uploading the firmware you have to go to console:
1). Type in PULSETIME 15 (or any number your relay shield wild then be push button)
2). 2a. type: switchmode 2
2b. type: switchtopic alarm

Thank you for that one as well. There is so many features in this firmware that it keeps on amazing me!!

bandric1, 2017-11-06:
has anyone encountered an issue when wenos is being trigrerred to open door by itself i checked logs once the wifi was lost onve it was a wemos reboot once it happened when mosquito reboot. Does anyone have an idea how to stop this from happening.

smulle48, 2017-11-25:
@bandric1 I have the exact same problem. When my wemos d1 mini is connected to my mqtt server, it all work as it supposed to, but if the demos loose connection to the mqtt server, my reed switch, act like a real switch, switching the relay.

smulle48, 2017-11-26:
@bandric1 I finally got it working. Switch1 will always act like a real switch for relay1 if no mqtt connection. I have made mine with a switch1 not connected to anything. I have then added a reed switch to switch2 and send the command: "switchmode2 2" and "switchtopic2 alarm". now the reed switch return "stat/sonoff/RESULT = {"POWER2":"OFF"}" and "stat/sonoff/RESULT = {"POWER2":"ON"}" without toggle the relay1 :-)

bandric1, 2017-11-27:
Great please post screenshots from tasmota web interface and settings from things, items and sitemap. So everyone can benefit.

smulle48, 2017-11-27:
I use the Wemos D1 mini with relay shield.

The relay is connected to D1 (GPIO5)

I have set it up to:
D1 (GPIO5) = Relay1
D6 (GPIO12) = Switch2 (Reed switch)
D8 (GPIO15) = Relay2

The I console I ran the command:
PulseTime 10
Switchmode2 2
SwitcoTopic2 alarm

Now Relay1 (relay shield) is independent from switch2 (reed switch) even if there is no mqtt connection. When you look at the main menu, there is actually 2 relay and the reed switch is controlling the relay2 which have no physical connection.

Now when firing the command cmd/sonoff/POWER1 ON the relay turn on for about 1 sec and the turn off again.

The status of POWER2 (reed switch) I am catching in node-red to fire a pushover message to my iPhone telling if the garage door is open or closed.

[image]

bandric1, 2017-12-15:
I finally got some time tried it u used now latest tasmota firmware but here is something with this firmware that it doesnt like this wemos d1 mini after mosquito restart or device looses wifi or restarts after power loss garage door opens im loosing it i guess i will just give up on this and try other firmware since this is not working for me.

bandric1, 2017-12-15:
all my other devices sonoff and esps dont supper from this only the wemos d1 mini does

smulle48, 2017-12-15:
Have you set it to:

PowerOnState 0

If it’s set to on, it will toggle the relay on restart

davidelang, 2017-12-15:
When the ESP8266 is first enabled, the GPIOs go to a known state before any 
software gets to do anything to them.

Relays can be wired to be on when the GPIO is low, or when it's high.

If you are having problems with the relays momentarily switching to the wrong 
state before going to the correct state via poweronstate, then you need to get a 
module that has the relays wired the other way (and in the module config, you 
change from relay1 to relay1inv, the inv being for inverted)

for the types of things the sonoffs were designed for, this momentary pulse in 
the wrong state doesn't hurt anything, but for opening the garage door, this can 
be enough to trigger problems.

bandric1, 2017-12-15:
I changed the wemos d1 module with other one and now it works. Damn i used so much time on this.
But now when i put a pin D6 to GND the RELAY2 doesnt swicht/change from OFF to ON.

bandric1, 2017-12-15:
It works as meant thank you guys.

JuliusLedoux, 2018-03-18:
Hi There I hope that someone can help me with this situation: I have setup a sonoff sv as my garage controller. My configuration is:
[image]
I have setup the reed switch on GPIO14 as switch2 and I also have GPIO5 being used for a DHT22 temperature sensor.
I have done the following configuration steps on the console:
pulsetime 10 (To Turn the relay off after one second), 
switchmode1 0,  then switchmode2 1 and then I have done swithctopic2 DoorStatus.
I am using Home Assistant as my HUB. In my subscription topic I am using the following:
cover:
  - platform: mqtt
    name: "Garage Door"
    state_topic: "cmnd/DoorStatus/POWER2"
    command_topic: "cmnd/garage/POWER"
    availability_topic: "tele/garage/LWT"
    payload_open: "ON"
    payload_close: "ON"
    payload_stop: "ON"
    payload_available: "Online"
    payload_not_available: "Offline"
    state_open: "ON"
    state_closed: "OFF"
    optimistic: false
    retain: false
I also have an automation that alerts me when the door is open; everything seems to be working fine. I can open and close my garage from the Application and I can see the status changing accordingly, on the screen, but some reason that I do not understand my automation tells me that the door is open (again) when the door it is just closed. It seems to be generating 2 events on each situation.
For what is worth this is my automation:
  alias: Play Media When Garage Opens
  trigger:
  - entity_id: cover.garage_door
    platform: state
    from: closed
    to: open
  condition:
  - condition: state
    entity_id: group.family
    state: home
  action:
  - data:
      message: Heads up! The Garage is Open
    entity_id:
    - media_player.living_room_speaker
    service: tts.google_say

When I look at the logs on Tasmota this is what I see:

**Opening:**
16:09:59 RSL: Received Topic cmnd/garage/POWER, Data Size 2, Data ON
16:09:59 RSL: Group 0, Index 1, Command POWER, Data ON
16:09:59 MQT: stat/garage/RESULT = {"POWER":"ON"}
16:09:59 MQT: stat/garage/POWER = ON
**16:10:00 MQT: cmnd/DoorStatus/POWER2 = ON
16:10:00 MQT: cmnd/DoorStatus/POWER2 = OFF
16:10:00 MQT: cmnd/DoorStatus/POWER2 = ON**
16:10:00 MQT: stat/garage/RESULT = {"POWER":"OFF"}
16:10:00 MQT: stat/garage/POWER = OFF

**Closing:**

16:12:25 RSL: Received Topic cmnd/garage/POWER, Data Size 2, Data ON
16:12:25 RSL: Group 0, Index 1, Command POWER, Data ON
16:12:25 MQT: stat/garage/RESULT = {"POWER":"ON"}
16:12:25 MQT: stat/garage/POWER = ON
16:12:26 MQT: stat/garage/RESULT = {"POWER":"OFF"}
16:12:26 MQT: stat/garage/POWER = OFF
16:12:33 WIF: Checking connection...
16:12:33 WIF: Connected
**16:12:35 MQT: cmnd/DoorStatus/POWER2 = OFF
16:12:35 MQT: cmnd/DoorStatus/POWER2 = ON
16:12:35 MQT: cmnd/DoorStatus/POWER2 = OFF**

Any help on this is highly appreciated!

dastrix80, 2018-05-02:
Does anyone know in terms of sensors if this would work with the above code?

https://www.ebay.com.au/itm/2Pcs-Alarm-Magnetic-Reed-Switch-Detector-Sensor-Roller-Shutter-Garage-Door/201765324857?epid=2121013235&hash=item2efa268839:g:ar0AAOSwa~BYZNTO

JuliusLedoux, 2018-05-02:
I believe it will work. Just verify if it is Normally open or Normally close.

dastrix80, 2018-05-02:
I shall do. I will need a Normally Closed sensor.

I also have a seperate relay free on my Sonoff 4CH WIFI Pro R2, I'm assuming I can use this and not GPIO14 on the Sonoff Basics as per this thread. I have tasmota on there. Do you know whats requird Julius to have it function?

JuliusLedoux, 2018-05-02:
I do not have  a Sonoff 4CH. But based on the specs it should  work in a similar way that it does on the basic one. Take a look   at  https://youtu.be/uwFmfC0iJiY

i3laze, 2018-07-10:
Here's how I just did **without** MQTT (Sonoff Basic, Reel sensor on Gpio 14 + Gnd):

Module Configuration:
- still Sonoff Basic
- GPIO4 - relay2 _- virtual relay which will hold the state of the reel_
- GPIO14 - switch2

Switchmode2 2 _(for NO mode, 1 for NC mode) - this way Relay2 status will follow the state of the reel correctly
Pulsetime1 0.5 _- in my case I use pulse for controlling garage door unit circuit._

Now it's time to write a HomeBridge plugin, which will control the whole logic.

jcconnell, 2018-07-11:
@i3laze can you explain the virtual relay concept a bit more? I've configured two Sonoff basics for use as garage door openers, with a reed switch on GPIO14. I have them setup in Home Assistant, but when Home Assistant is rebooted, the state of the garage door is not shown. 

I could set the powerrtain on but then relay1 is cycled on each reboot. I'm wondering if having the virtual relay to store the state could nullify this shortcoming, so that I can still track the state of the garage across reboots.

ascillato, 2018-07-11:
Hi @jcconnell

To make home assistant read the status of Tasmota when home assistant restarts, it is explained at the wiki in the home assistant part . It is just adding an automation to make all sonoffs inform its status when home assistant starts.

ascillato, 2018-07-11:
At Tasmota wiki I meant

jcconnell, 2018-07-11:
It looks like that is sending a command to the sonoff. How do I read the state of the sensor?

The statetopic for the garage door reed sensor is "cmnd/garage_door_status/POWER2"

ascillato, 2018-07-11:
If you want that Home Assistant know the state of all your Tasmota devices when Home Assistant Starts, just copy and paste (without modifying anything) to your configuration.yaml the config explained in the wiki at

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#power-state-on-start-up-without-retain-option

No need to use retain values in Home asistant neither in the MQTT broker. The internal state of the Tasmota need to be just in Tasmota.

jcconnell, 2018-07-12:
Awesome, that works well for my other sonoffs. Thank you. I have a unique use case for the two I'm using as garage door openers though. I need to keep track of the status of a reed sensor connected to GPIO14 on these two Sonoffs. The reed sensor indicates the position of the garage door: Normally closed so 1=open, 0=closed. The relay controls the actuation of the door: 1 sec pulse = pressing the wall mounted button.

i3laze, 2018-07-12:
Actually, I don’t like all these workarounds at all. 

There in Tasmota should be a separate “sensor” role for reeds/other NC/NO stuff.
So you could just query status, like “Power” does for relays.

jcconnell, 2018-07-12:
I noticed that the telemetry reports the state of the reed switch on GPIO14. I wonder if it's possible to use that instead?

ascillato, 2018-07-12:
Yes, the telemetry is intended for that. You can use that of course.

You should have a telemetry like:

`tele/sonoff/SENSOR = {"Time":"2018-07-12T12:00:05","Switch1":"ON"}`

So, put in your configuration.yaml

```yaml
sensor:
  - platform: mqtt
    name: "Humedad Ambiente"
    state_topic: "tele/sonoff/SENSOR"
    value_template: "{{ value_json['Switch1'] }}"
```

i3laze, 2018-07-12:
> tele/sonoff/SENSOR = {"Time":"2018-07-12T12:00:05","Switch1":"ON"}
Any web-command to query that without MQTT?

jcconnell, 2018-07-12:
Here is what I have currently:

```cover:
        - platform: mqtt  
           name: "Garage Door Sonoff" 
           state_topic: "cmnd/garage_door_status/POWER2"
           command_topic: "cmnd/garage_door/POWER"
           availability_topic: "tele/garage_door/LWT"
           payload_open: "ON"
           payload_close: "ON"
           payload_stop: "ON"
           payload_available: "Online"
           payload_not_available: "Offline"
           state_open: "ON"
           state_closed: "OFF"
           optimistic: false
           retain: false
```
Would this be the correct implementation to use telemetry as state?
```cover:
        - platform: mqtt  
           name: "Garage Door Sonoff" 
           state_topic: "tele/garage_door_status/SENSOR"
           value_template: "{{ value_json['Switch2'] }}"
           command_topic: "cmnd/garage_door/POWER"
           availability_topic: "tele/garage_door/LWT"
           payload_open: "ON"
           payload_close: "ON"
           payload_stop: "ON"
           payload_available: "Online"
           payload_not_available: "Offline"
           state_open: "ON"
           state_closed: "OFF"
           optimistic: false
           retain: false
```

Finally, I've noticed that if I reboot a Sonoff, the item becomes unavailable in HA, but it never becomes available again once the reboot finishes. Any idea why?

ascillato, 2018-07-12:
In your last config you are mixing controlling with sensing.

Those need to be separated.

n00bster-za, 2022-05-04:
has been a while since someone posted here. I need assistance on getting the GPIO14 to read more than just on or off. I have a gate motor with a dedicated Gate status output. The gate status can be open,close,opening,closing,collision. How do I configure the GPIO to read all those different states and add it to HA.

ascillato2, 2022-05-04:
Hi, this issue is very old and your question is not directly related to this. Please, open a new discussion posting all the information you have (hardware type, tasmota version, what have you tried, etc), or ask in the support chat. Thanks.