arendst/Tasmota #727: Secured and Easy ESP8266

liorsu, 2017-08-12:
Hello,

I have an esp8266 with 1MB flash.
I want to be able to use it with two relays and a one DHT22.
I do want it to communicate over tls with an mqtt server in the internet, but still have the web server for initial and easy configuration.
Is it possible to use USE_MQTT_TLS and USE_WEBSERVER while disabling other redundant features (such as USE_I2C, USE_IR_REMOTE, USE_DOMOTICZ, etc) and have it working?
As far as for firmware.bin size after compiling I got it to be 497kb, 500kb is the limit right?

liorsu, 2017-08-12:
Forgot to mention that I commented the assertion
#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  //#error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

liorsu, 2017-08-12:
I think I just succeeded in doing that (with a sonoff sv upgrade to 4MB flash).
Can the 1MB flash be the problem with a regular sonoff?
Few more questions...
1) I would really appreciate it if it will be possible to add an option to edit the tls finger print from the web server
2) I looked into the code and saw that besides printing a warning, the TLS validation does not effect much:
    if (espClient.verify(sysCfg.mqtt_fingerprint, sysCfg.mqtt_host)) {
      addLog_P(LOG_LEVEL_INFO, PSTR("MQTT: Verified"));
    } else {
      addLog_P(LOG_LEVEL_DEBUG, PSTR("MQTT: Insecure connection due to invalid Fingerprint"));
    }
Is it also possible to add a define to choose if this is the wanted behavior or just to drop the connection as we are possibly connected to an MITM server?

davidelang, 2017-08-13:
what is the size of the compiled image that you are getting as a result of your 
options?

liorsu, 2017-08-13:
As I wrote above, 497kb

davidelang, 2017-08-13:
so what happens when you attempt to flash it? are you sure you're using DOUT 
mode?

liorsu, 2017-08-13:
For the esp8285 with the flashes tasmota firmware I tried to upload them mqtt tls+web version via the upgrade interface, but it just restarted with the previous version.
And the sonoff sv (esp8266) with the 4MB upgrade, it worked fine. Just didn't really care much about the tls fingerprint verification, and I also couldn't change it from the web as stated above. Only by flashing..

davidelang, 2017-08-13:
there was a break early in the 5.x series that broke updating from versions 
prior to 5.0, you have to upgrade in two steps, first to the latest 5.x (5.11 or 
so IIRC) that could upgrade via a 4.x firmware, and then upgrade from that to 
current.

but if you are upgrading via serial, then this doesn't matter.

liorsu, 2017-08-13:
Both version are 5.x
Just one without tls and the other with.
I don't really know why but I can't get the eap8285 to flash mode again to flash it via serial.. so was trying via the web upgrade mechanism

davidelang, 2017-08-13:
There is nothing that can be done in software to disable the ability to flash 
via serial, if you are connected properly and ground GPIO0 before applying power 
to the device, it WILL go into flash mode and let you flash it.
k

closed 2018-05-07 by stale[bot]