arendst/Tasmota #728: Problem upgrading to v5 with ubuntu and platformio - error 255

anokinuk, 2017-08-12:
Hi,
I am using ubuntu and platformio trying to update the current Sonoff-Tasmota version to a previously flashed sonoff 4ch (ver 4.x). Each time I upload I get 
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/sonoff-minimal/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
I had no problem with previous versions. Apologies if this is an error on my part. I have tried to find a solution but Google has not been my friend so far. Any guidance appreciated.

davidelang, 2017-08-13:
updating via serial means that it doesn't matter what's on there before.

If you aren't able to flash, you have something setup wrong, either reversed 
tx/rx, something else keeping the serial port open, incorrect baud rate, or 
similar problem.

anokinuk, 2017-08-14:
Thanks for you help. I have tried to swap rx, and tx (tried both ways) but I get the same result

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
Apologies but I not an expert at this, so any help appreciated.

davidelang, 2017-08-14:
If you don't have the right port, you won't be able to flash it.

what OS are you using? are you flashing through the arduino IDE? (that's 
probably the best option if you don't know what you're doing)

anokinuk, 2017-08-26:
Thanks for your help. It turned out to be a dodgy bit of soldering on the GP0 switch. Working fine now.

closed 2017-08-26 by anokinuk