arendst/Tasmota #730: device bricked with 5.5.2b - blue LED is still blinking

andreasloe, 2017-08-13:
I am using 5.5.2b and managed to compile the software on a MacBook. Prerequisites (DOU etc.) were also taken care of. The POW was working when it arrived from China.

I never managed to copy the firmware to the device, I usually got the error

Arduino: 1.8.1 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), ck, Disabled, None"

Der Sketch verwendet 483420 Bytes (47%) des Programmspeicherplatzes. Das Maximum sind 1023984 Bytes.
Globale Variablen verwenden 43196 Bytes (52%) des dynamischen Speichers, 38724 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.

/Users/xxxxx/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd ck -cb 115200 -cp /dev/cu.usbserial-A505BSWS -ca 0x00000 -cf /var/folders/l6/myrckv8d12z9_06fl49fm36h0000gn/T/arduino_build_836398/sonoff.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A505BSWS
	setting address from 0x00000000 to 0x00000000
Beim Hochladen des Sketches ist ein Fehler aufgetreten
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A505BSWS at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
<this comes several times, untill>
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

I happened to have two different USB-TTL adapter (one from a Chinese vendor, the other was a arduino nano with GND-REST connected). My device does not react anymore, so maybe it is bricked. 

With the help of a friend  tried to activate at least the bootloader (opening serial monitor, connecting 3.3V and listening to the monitor) - unfortunately, nothing appeared.

What I do notice is the following: If I connect the device to main (all cables from arduino removed!!!) the blue LED goes on for about 15 seconds or so, goes off very shortly and goes on again 15 sec. Pressing the button does nothing. Is the POW really broken or is there something I can do?

And if it is broken what parts do you suggest should be reused (I hate throwing things away...). Thank you!

TD-er, 2017-08-15:
Just to be sure. Please also check the supplied voltage of the USB-adapters you have connected to the Sonoff.
Some of them do not regulate the voltage on the Vcc pin down to 3.3V and only lower the signal levels when switching to 3.3V That means the 5V from the USB is connected to the Sonoff and that's what they don't like. Some may survive just enough to be flashed once, but die eventually.

andreasloe, 2017-08-16:
I am sorry for this open issue - it was my fault. I simply connected Tx to Tx and Rx to Rx, which is stupid. After connecting Transmitter and Receiver correctly everything worked! (I also checked the voltage beforehand.)

Maybe we should comment anywhere in the Wiki that an errormessage  
`warning: espcomm_sync failed
`error: espcomm_open failed
`error: espcomm_upload_mem failed

is insofar useless because it does not tell the source of the error. One should always check the "full debug while uploading" checkbox in the IDE.

closed 2017-08-16 by andreasloe