arendst/Tasmota #731: Failed to compile on visual micro (visualStudio)

gnkarn, 2017-08-13:
i understand it tis not the recommended environment from the wiki, but as it is the one i use on daily bases , im trying to make it work 

this is the error log at compile time for the sonoff.ino file . All libraries seams to be correctly identified and linked , but ..

would appreciate if there is some advice of what else i could try .

thank you
gustavo
-----------
, 
Building project code ...
"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_ESP01\" -DESP8266 -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" -I"C:\Users\gnkarn\Documents\Arduino\libraries\PubSubClient\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate\src" -I"C:\Users\gnkarn\Documents\Arduino\libraries\ArduinoJson\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer\src" -I"C:\Program Files (x86)\Arduino1.8\arduino-1.8.1\libraries" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries" -I"C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\2iftqv1b.p1s\Micro Platforms\default\debuggers" -I"C:\Users\gnkarn\Documents\Arduino\libraries" "C:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release\sonoff.cpp" -o "C:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release\sonoff.cpp.o"
 
sonoff.ino: 106:20: error: variable or field 'IPtoCharArray' declared void
   void IPtoCharArray(IPAddress address, char *ip_str, size_t size)
 
sonoff.ino: 106:20: error: 'IPAddress' was not declared in this scope
 
sonoff.ino: 106:39: error: expected primary-expression before 'char
   void IPtoCharArray(IPAddress address, char *ip_str, size_t size)
 
sonoff.ino: 106:60: error: expected primary-expression before 'size
   void IPtoCharArray(IPAddress address, char *ip_str, size_t size)
 
sonoff.ino: 119:36: error: 'TIME_T' has not been declared
   void breakTime(uint32_t timeInput, TIME_T &tm)
 
sonoff.ino: 120:19: error: 'TIME_T' was not declared in this scope
   uint32_t makeTime(TIME_T &tm)
 
sonoff.ino: 120:27: error: 'tm' was not declared in this scope
   uint32_t makeTime(TIME_T &tm)
 
sonoff.ino: 121:19: error: 'TimeChangeRule' was not declared in this scope
   uint32_t toTime_t(TimeChangeRule r, int yr)
 
sonoff.ino: 121:37: error: expected primary-expression before 'int
   uint32_t toTime_t(TimeChangeRule r, int yr)
 
sonoff.ino: 121:43: error: expression list treated as compound expression in initializer [-fpermissive]
   uint32_t toTime_t(TimeChangeRule r, int yr)
 
sonoff.ino: 133:31: error: 'IPAddress' has not been declared
   void startWebserver(int type, IPAddress ipweb)
 
sonoff.ino: In function String getUTCDateTime()
 
support.ino: 801:29: error: invalid initialization of reference of type 'int&' from expression of type 'TIME_T
   breakTime(utctime, tmpTime)
 
sonoff.ino: 119:6: error: in passing argument 2 of 'void breakTime(uint32_t, int&)
   void breakTime(uint32_t timeInput, TIME_T &tm)
 
sonoff.ino: In function uint32_t makeTime(TIME_T&)
 
support.ino: 867:29: error: 'uint32_t makeTime(TIME_T&)' redeclared as different kind of symbol
   uint32_t makeTime(TIME_T &tm)
 
sonoff.ino: 120:10: error: previous declaration of 'uint32_t makeTime
   uint32_t makeTime(TIME_T &tm)
 
sonoff.ino: In function uint32_t toTime_t(TimeChangeRule, int)
 
support.ino: 898:43: error: 'uint32_t toTime_t(TimeChangeRule, int)' redeclared as different kind of symbol
   uint32_t toTime_t(TimeChangeRule r, int yr)
 
sonoff.ino: 121:10: error: previous declaration of 'uint32_t toTime_t
   uint32_t toTime_t(TimeChangeRule r, int yr)
 
support.ino: 921:20: error: 'makeTime' cannot be used as a function
   t = makeTime(tm);        \\ First day of the month, or first day of next month for "Last" rules
 
sonoff.ino: In function void rtc_second()
 
support.ino: 985:45: error: 'toTime_t' cannot be used as a function
   dsttime = toTime_t(myDST, rtcTime.Year)
 
support.ino: 986:45: error: 'toTime_t' cannot be used as a function
   stdtime = toTime_t(mySTD, rtcTime.Year)
"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release/core" -c -w -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_ESP01\" -DESP8266 -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" -I"C:\Users\gnkarn\Documents\Arduino\libraries\PubSubClient\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate\src" -I"C:\Users\gnkarn\Documents\Arduino\libraries\ArduinoJson\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer\src" -I"C:\Program Files (x86)\Arduino1.8\arduino-1.8.1\libraries" -I"C:\Users\gnkarn\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries" -I"C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\2iftqv1b.p1s\Micro Platforms\default\debuggers" -I"C:\Users\gnkarn\Documents\Arduino\libraries" "C:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release\core_esp8266_timer.c" -o "C:\Users\gnkarn\AppData\Local\Temp\VMBuilds\sonoff\esp8266_generic\Release\core_esp8266_timer.c.o"
Error compiling project sources
Build failed for project 'sonoff'

gnkarn, 2017-08-13:
using visual studio with visual micro and arduino IDE 1.8.1
selected 1Mb generic esp , with no SPIFF

mkh595, 2017-08-14:
Who once tried VisualStudio will never return to ArduinoIDE because it is terrible.

It is not compiled in VS because it required more strict C++ convention (e.g. declaration in header files) which Tasmota code doesn't respect (use .ino files). Arduino IDE use different way how to build libraries and code separated in many .ino files than VS.

Therefore I had to install platformio for VisualStudioCode and do compilation there.

gnkarn, 2017-08-14:
yes , i know what you mean ,i managed on many other projects  to compile on visual studio, but this particular project is being difficult , as  most of the error messages  gives no useful information to help  during debug .
thanks anyway.

davidelang, 2017-08-14:
well, that means that you are bypassing all the easy ways to do things and 
trying the hard way, it's no wonder you are running into grief.

Since there is virtually no editing that you are needing to do, I would suggest 
that you try compiling/flashing with the arduino IDE (even if you do all your 
editing with other software, like I usually do)

davidelang, 2017-08-14:
one of the reasons visual studio is going to have so much trouble is that 
Tasmota doesn't have one top-level file and then include everything else from 
there. It's written around the Arduino IDE 'feature' that it will grab all .ino 
files in the directory and compile them all, even without includes.

gnkarn, 2017-08-15:
Ok, I am now facing other bigger trouble , because everything I flash to the sonoff ends ok, but do not run , it only throws the exception 0 , error on terminal .
And is independent of the sketch flashed, even blink do not work ...
I saw there is a lot of people facing the same problem , with no clear solution identified , 
So , spending lot of time on this angle of research now ...

mkh595, 2017-08-15:
```
well, that means that you are bypassing all the easy ways to do things and
trying the hard way, it's no wonder you are running into grief.

```

That's no true. With VS we can do programing easily and more evectively than in primitive IDE editor.
Can you navigate in IDE to declaration or subroutine or to library with one click? Do you see all changed lines marked? Can you jump to place with error just with click on error message? Display IDE tooltip with parameters when you add new function or structure? 
Those and other features make programing more comfortable so `hard way` is not true.

I did all my arduino projects (also splitted in several files) in VisualStudio, only Tasmota is writen in way that cannot be compiled.

davidelang, 2017-08-15:
you are mixing up different aspects of things.

I don't at all dispute that Visual Studio is better for writing programs than 
the Arduino IDE.

but we're not talking about writing programs, we are talking about supporting 
different libraries and cross compilers and then flashing the devices. The 
Arduino IDE does 'magic things' to make it easier for the 'non programmers' who 
are the arduino target audience. These things violate standard C programming 
approaches, which is why doing it without using the arduino IDE to compile and 
flash is doing those tasks 'the hard way'.

I am not a fan of the way Arduino does things, or the fact that so many people 
are jumping on the 'make it work in the Arduino IDE' bandwagon.

But it doesn't really matter that I'm not a fan, this stuff is out there and we 
need to either work with it, or re-write it :-)

When you post that you are a novice at working with devices like this (as you 
described yourself), you should go ahead and use whatever you want for editing 
the files, but then fire up the Arduino IDE to compile/flash the software until 
you are comfortable with the process of flashing. Then you are in a far better 
position to learn the command-line version of things where you have to do a 
bunch of things differently.

I tend to edit the files using vi in one of the many xterms that I have open on 
any one screen, and then fire up the arduino IDE just long enough to compile and 
flash.

gnkarn, 2017-08-15:
probably i was not clear enough , ( vocabulary problem from my english most probably) ,
i fully agree with the IDE, i have no problems with Arduino IDE , i always preferred to work with VS , etc 

but ,
what im saying , is that there is a problem , with sonoff devices , and they tend to get people into trouble, and they behave like being bricked , after reflashing ( using whatever means) , and im facing that problem now , so , i fully understand that this has nothing to do with the IDE, i found many comments from people facing exactly the same problem , but no one seams to have a solution , this morning i found something to test on this thread : https://www.letscontrolit.com/forum/viewtopic.php?t=2772&start=10 , so , will see ..

tks

davidelang, 2017-08-15:
It is not possible to brick an ESP8266 (you can destroy it with excess voltage 
easily, but not brick it)

The software in the chip that puts it in flash mode when you ground GPIO before 
applying power CANNOT be changed.

If you are having problems flashing via serial you have one of a few problems

1. you are not grounding GPIO before power is applied

2. you are not wired to the device correctly (tc/rx reversed, etc)

3. you are doing something wrong in software.

if you are facing problem #3, using software that requires many more options 
being manually set to make it work instead of software that is easier to use 
(and that more people are familiar with) is not a great move.

This sub-thread started with a couple posts that (paraphrased) said "I don't 
know anything about the command line or the software I'm using, but I'm going to 
refuse to consider using the software that most people use"

It's fine to try things, but when you run into trouble, you need to make sure 
that you are fighting the right problem. So try using the software that does 
more stuff for you and has fewer ways to do things wrong. This lets you figure 
out all the issues in categories #1 and #2. Once you are comfortable with those,
then you can try the different software options.

gnkarn, 2017-08-15:
thanks david for your comments .

im not having problems with flashing
the esp is acting as  if the sketch were flashed normally, and it ends ok, so  not 1) , not 2) problems
and obviously not 3) because i did it many times, and have been testing with proven, tested , easy sketches

so, at this point i suspect that the problem is the sonoff itself , 
i would understand if it were " totally failed" , but partially fail ???, it is hard to believe , as it was working in the first place , i can erase, it ,  i can check it , using esptool.py , etc 

but

after loading , any sketch , with any tool , the result is always the same : 

ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
Fatal exception (0): 
epc1=0x4021c1d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

so,  no progress at all ..

davidelang, 2017-08-15:
it's also possible that I've mixed up threads (replying via e-mail and then 
deleting the message makes that easier to do)

As you say, if you are flashing and no matter what you flash, it gives the same 
result, that sounds like a failed chip.

'partially failed' is actually not that common, if 5v is applied to the main 
power of the ESP, the entire thing will blow, but if you managed to apply too 
much votage to one of the other pins, it's very possible for the chip to only 
partially fail (I've seen it reported before)

turn on detailed logging of the flashing session and triple check that there are 
no errors being thrown during the flashing process. If not, then I'd write off 
that module and pick up another one.

At least they're cheap

gnkarn, 2017-08-15:
agree, will try with other sonoff ,
and see what happens , is just that feeling , that when you dont have a  probable cause identified, is that  there are BIG chances, that  it may happen again …

voltage , is not the case , as im working with all 3,3 volts at this moment .

will see, 
thanks

mkh595, 2017-08-16:
@davidelang 

`When you post that you are a novice at working with devices like this (as you
described yourself),`

Please don't talk nonsense. I never write this.

davidelang, 2017-08-16:
On Wed, 16 Aug 2017, mkh595 wrote:

> @davidelang
>
> `When you post that you are a novice at working with devices like this (as you
> described yourself),`
>
> Please don't talk nonsense. I never write this.

my apologies then, I must have mixed you up with a different thread complaining 
about difficulties flashing where the person wasn't willing to use the Arduino 
IDE because Visual Studio was so much better as a programming environment.

PeteBa, 2017-08-19:
So, I had this problem also. As already mentioned, Arduino IDE uses a different build logic than Visual Micro. The VM author recognises this but has other priorities on his plate so unlikely to see support anytime soon. I ended up writing two powershell scripts to build and upload using the Arduino IDE compatible logic.

I call these scripts from within Visual Studio rather than using the equivalent Visual Micro commands. I use custom ribbon buttons to call the scripts using the External Tool functionality . This gives me the benefit of the VS IDE but with the ability to build/upload Sonoff-Tasmota.

It's far from perfect as you need to hard-code the Visual Micro settings (i.e. CPU Frequency, Flash Mode, COM Port ...) into the scripts but it does at least support click-through on build error messages in the Output window. It doesn't auto-save files before doing a build/upload so you need to save any modified files first.

In case anybody is interested I've attached here. Let me know if useful.
[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/1236475/sonoff.zip)

closed 2018-05-07 by stale[bot]