arendst/Tasmota #732: Fail to compile on IDE because MQTT library?

J-Ubeda, 2017-08-14:
When compiling with IDE (by example sonoff.ino) i get this error:

D:\Usuarios\BAUBMOFR\Documents\Arduino\Sonoff-Tasmota-development\sonoff\sonoff.ino:160:52: fatal error: PubSubClient.h: No such file or directory

 #include <PubSubClient.h>                   // MQTT

It seems some library is not loaded? What ? Where find it?

ionciubotaru, 2017-08-14:
please read the .... manual [https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite)

J-Ubeda, 2017-08-15:
Thanks, problem solved.

closed 2017-08-16 by arendst

jeylites, 2017-09-09:
J-Ubeda,

I'm having the same problem, "PubSubClient.h: No such file or directory" what did you do to fit it?

J-Ubeda, 2017-09-11:
My problem was i did not load the corresponding library.

The library is this:

[cid:a1e0bf3a-f741-441c-9302-8b2e452b1ef7]
[cid:9989d2f7-73a8-4d63-8dba-e80d21c247de]

vmarks, 2018-10-26:
attempting to compile with arduino. I click through the prerequisites, arduino. It doesn't mention loading a library, or where to get the PubSubClient prerequisite.

ascillato, 2018-10-26:
please, read the wiki. all the instructions are there

vmarks, 2018-10-26:
If you follow the steps on the Arduino page of the wiki, they aren't.

https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE - try it. There's nothing here or on Troubleshooting that answers compile errors like this using Arduino. Where should I be looking instead?

ascillato, 2018-10-27:
Yes it is,

Sorry before I was with my phone and couldn't paste a direct link for you.

your answer is exactly at https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#for-all-versions

vmarks, 2018-10-27:
Follow-up - I had been using source I downloaded two nights ago. I downloaded fresh development source and it built. No idea why.

vmarks, 2018-10-27:
I'm not sure you saw, but you and I linked to the same page. I followed the instructions there, and it wouldn't build. It didn't address the PubSubClient.h issue at all on that page. Thanks for the link and trying to help me. I'm making slow progress.

vmarks, 2018-10-27:
Thank you. I wish the authors had not presumed those paths were already known. I wish I were more familiar with Arduino so I would have seen that and understood where it wanted those files copied.
Here's where I'm at. 

I still get build errors in arduino IDE, although they're different errors now. `Arduino: 1.8.7 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, QIO, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 57600"

In file included from /Users/vmarks/Documents/Arduino/sonoff/sonoff.ino:36:0:
sonoff_post.h:42:30: error: variable or field 'KNX_CB_Action' declared void
 void KNX_CB_Action(message_t const &msg, void *arg);
                              ^
sonoff_post.h:42:20: error: 'message_t' was not declared in this scope
 void KNX_CB_Action(message_t const &msg, void *arg);
                    ^
sonoff_post.h:42:42: error: expected primary-expression before 'void'
 void KNX_CB_Action(message_t const &msg, void *arg);
                                          ^
In file included from /Users/vmarks/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34:0,
                 from /Users/vmarks/Documents/Arduino/sonoff/sonoff.ino:49:
/Users/vmarks/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h: In member function 'bool ESP8266WiFiSTAClass::beginWPSConfig()':
/Users/vmarks/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h:100:103: error: 'WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool' was not declared in this scope
             return WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool();

                                                                                                       ^
/Users/vmarks/Documents/Arduino/sonoff/xdrv_05_irremote.ino: In function 'boolean IrSendCommand()':
xdrv_05_irremote:299:7: error: 'StaticJsonBuffer' was not declared in this scope
       StaticJsonBuffer<128> jsonBuf;
       ^
xdrv_05_irremote:299:29: error: 'jsonBuf' was not declared in this scope
       StaticJsonBuffer<128> jsonBuf;
                             ^
xdrv_05_irremote:300:7: error: 'JsonObject' was not declared in this scope
       JsonObject &root = jsonBuf.parseObject(dataBufUc);
       ^
xdrv_05_irremote:300:19: error: 'root' was not declared in this scope
       JsonObject &root = jsonBuf.parseObject(dataBufUc);
                   ^
/Users/vmarks/Documents/Arduino/sonoff/xdrv_07_domoticz.ino: In function 'boolean DomoticzMqttData()':
xdrv_07_domoticz:168:5: error: 'StaticJsonBuffer' was not declared in this scope
     StaticJsonBuffer<400> jsonBuf;
     ^
xdrv_07_domoticz:168:27: error: 'jsonBuf' was not declared in this scope
     StaticJsonBuffer<400> jsonBuf;
                           ^
xdrv_07_domoticz:169:5: error: 'JsonObject' was not declared in this scope
     JsonObject& domoticz = jsonBuf.parseObject(XdrvMailbox.data);
     ^
xdrv_07_domoticz:169:17: error: 'domoticz' was not declared in this scope
     JsonObject& domoticz = jsonBuf.parseObject(XdrvMailbox.data);
                 ^
xdrv_07_domoticz:189:25: error: 'color' was not declared in this scope
             JsonObject& color = domoticz["Color"];
                         ^
/Users/vmarks/Documents/Arduino/sonoff/xdrv_09_timers.ino: In function 'boolean TimerCommand()':
xdrv_09_timers:364:11: error: 'StaticJsonBuffer' was not declared in this scope
           StaticJsonBuffer<256> jsonBuffer;
           ^
xdrv_09_timers:364:33: error: 'jsonBuffer' was not declared in this scope
           StaticJsonBuffer<256> jsonBuffer;
                                 ^
xdrv_09_timers:365:11: error: 'JsonObject' was not declared in this scope
           JsonObject& root = jsonBuffer.parseObject(dataBufUc);
           ^
xdrv_09_timers:365:23: error: 'root' was not declared in this scope
           JsonObject& root = jsonBuffer.parseObject(dataBufUc);
                       ^
/Users/vmarks/Documents/Arduino/sonoff/xdrv_10_rules.ino: In function 'bool RulesRuleMatch(byte, String&, String&)':
xdrv_10_rules:189:3: error: 'StaticJsonBuffer' was not declared in this scope
   StaticJsonBuffer<1024> jsonBuf;
   ^
xdrv_10_rules:189:26: error: 'jsonBuf' was not declared in this scope
   StaticJsonBuffer<1024> jsonBuf;
[... 22 lines cut ...]
 uint8_t kRepeat[5] = {
                  ^
In file included from /Users/vmarks/Documents/Arduino/sonoff/xdrv_05_irremote.ino:74:0:
/Users/vmarks/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.h:26:16: error: 'kRepeat' has a previous declaration as 'const uint64_t kRepeat'
 const uint64_t kRepeat = UINT64_MAX;
                ^
/Users/vmarks/Documents/Arduino/sonoff/xplg_ws2812.ino: In function 'void Ws2812Gradient(uint8_t)':
xplg_ws2812:225:48: error: invalid types 'const uint64_t {aka const long long unsigned int}[uint8_t {aka unsigned char}]' for array subscript
   uint8_t repeat = kRepeat[Settings.light_width];  // number of scheme.count per ledcount
                                                ^
xplg_ws2812:235:37: error: invalid types 'const uint64_t {aka const long long unsigned int}[uint8_t {aka unsigned char}]' for array subscript
     if (kRepeat[Settings.light_width] > 1) {
                                     ^
Multiple libraries were found for "IRremoteESP8266.h"
 Used: /Users/vmarks/Documents/Arduino/libraries/IRremoteESP8266
 Not used: /Users/vmarks/Documents/Arduino/libraries/arduino_229722
exit status 1
variable or field 'KNX_CB_Action' declared void

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
I also tried PlatformIO, which built without issue on a fresh development source download. 
When I upload it to the ifan02, it uploads successfully, and then I get nothing in a serial terminal and it doesn't join Wi-Fi.

ascillato, 2018-10-27:
Well, The arduino issue is that the libraries are not copied in the correct folder. Some other of the errors are a incomplete or corrupt Arduino installation. You need to uninstall everything and install again. You have also the arduino portable option if you have trouble installing it. If you are installing using windows 10 store, please try to download directly from arduino.com
Also you can try an older version of arduino like 1.8.2 that works fine.

arduino config:
also you have set QIO and need to be DOUT as explained in the wiki
also you have set 57600 baud and need to be 115200 as explained in the wiki

So, with all those config errors, it is very difficult that your Tasmota works.

May be you need to start from skratch.

Try to flash the precompiled bins instead of compiling by yourself.

May be it is better for you to follow any of the youtube videos that you can find in the tutorials from the wiki.

Please, don't panic. At first it is a little difficult to learn all this new concepts, but then you will achieve this.

vmarks, 2018-10-27:
Thank you.

I flashed precompiled bins. The issue there was, I could not get the device to join Wi-Fi. I decided to build so that I could compile in my SSID and WPA2 passphrase.

ascillato, 2018-10-27:
Have you tried the last version 6.2.1.18 from development branch at http://thehackbox.org/tasmota/ ?

ascillato, 2018-10-27:
Are you using a router of the brand fritzbox ?
Are you using a wifi repeater ?

ascillato, 2018-10-27:
ok,

Then please try the precompiled bins last version 6.2.1.18 from development branch at http://thehackbox.org/tasmota/

vmarks, 2018-10-27:
Hi, 
I have tried a precompiled build made at 12:01 20181027. I have not yet been able to check its terminal output, but in terms of button presses and attempting to get it to join Wi-Fi, I am so far unsuccessful. I will check terminal output later today.

vmarks, 2018-10-27:
Follow up - firmware flashed, and it's booted and on my wifi. Thank you!

ascillato, 2018-10-27:
Great!, Thanks for reporting