arendst/Tasmota #733: Reduce power off after system restart

0n3-70uch, 2017-08-14:
Hello,

I'm searching for solutions to reduce the power offs. First: set `WifiConfig 4` to skip the restart after a wifi reconnect. Is there any solution to skip the power off after a MQTT broker reconnect/restart? For example something with the `retain` state?

Thanks for help.

Greetings
Fabi

ionciubotaru, 2017-08-14:
When sonoff restarts it sends 3 MQTT messages named INFO1, INFO2 and INFO3
Please publish the content of them.

0n3-70uch, 2017-08-14:
Ok I will do it in the evening...

ionciubotaru, 2017-08-14:
send also the result of `STATUS 0`

davidelang, 2017-08-14:
read the wiki page on poweronstart, it tells you how to find what is retained in 
your mqtt server and clear it.

0n3-70uch, 2017-09-04:
At the moment the devices working fine... I close this issue now. Thanks for the your help.

closed 2017-09-04 by 0n3-70uch