arendst/Tasmota #735: Sonoff SI7021 sensor on TH10

cputoaster, 2017-08-14:
I have a TH10 with an official SI7021 sensor connected. It works fine through ewlink with original Sonoff FW 2.0.4. I dont get any sensor appearing when flashing Tasmota 5.5.2.

I can see that others got this combo to work (here https://github.com/arendst/Sonoff-Tasmota/issues/96 ), but I somehow cannot. I did set gpio4 to I2C SDA and gpio14 to I2C SCL as mentioned in the wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration . A scan in the console shows nothing: {"I2Cscan":"No devices found"}

How does the single connected data wire (https://www.itead.cc/wiki/Sonoff_Sensor_Si7021) work with the two gpio ports anyway? Does anyone have this working on a TH10?

cputoaster, 2017-08-14:
ah, looking at the schematics of the sensor, there is an MCU present, so that may translate to some kind of 1-wire protocol.

bobo-t, 2017-08-17:
same issue here, I have a TH16 / Si7021, works with fw 2.0.4, it's not detected with Tasmota 5.5.2b (configured as per wiki). 
with web log level 3 I see in console after reboot:

SHT1X: Sensor did not ACK command
I2C: T/RH? found at address 0x40

i2cscan returns "No devices found"

davidelang, 2017-08-17:
did you go to the module definition and set the GPIOs correctly?

bobo-t, 2017-08-17:
my settings: module 1 , gpio14 5, gpio6 6

cputoaster, 2017-08-18:
Hmm, from the wiki I thought it would be gpio4 to I2C SDA and gpio14 to I2C SCL. But googling the cable connector pinout shows that only gpio14 is actually connected, so the protocol cannot be I2C as that would require two GPIOs.

arendst, 2017-08-18:
Well there is the bare SI7021 which is an I2C device and there is the ITEAD provided SI7021 external sensor which is the same I2C device controlled by an internal microcontroller to make it a one wire device.

Tasmota currently only supports the bare I2C device as I have no info about the one wire ITEAD protocol.

bobo-t, 2017-08-18:
I soldered a strap on TH16 PCB to make available GPIO4 to 4-wire jack and then i decoupled the SCA and SCL pins from MCU on ITEAD's Si7021 sensor and connected the wires to Si7021 chip (including the extra one), now it's working:

STATUS10 {"StatusSNS":{"Time":"2017-08-18T12:59:20", "SI7021":{"Temperature":28.8, "Humidity":37.1}, "TempUnit":"C"}}
 
anyway it's probably better to use 3-wire type of sensor .

chris-008, 2017-10-09:
I have the same issue with a non functioning Itead SI7021 sensor.
So just to be sure, If we buy an alternative single wire sensor as summarized at  https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration , we should be good to go? If so , any particular type which is recommended?

cputoaster, 2017-10-09:
AM2301 works great.

chris-008, 2017-10-10:
Thx cuptoaster, I'll buy one!

aamxxx, 2017-10-10:
I already bought two TH10 with two ITEAD's Si7021 (before reading this). Will Tasmota support them in the future or do I better buy supported ones?

jctual, 2017-10-13:
Does anyone know the protocol used on ITEAD 1-wire si7021 to be able to develop a plugin ?

cputoaster, 2017-10-16:
FYI, I asked on their forum. This is ITEADs answer:

"Sonoff Si7021 just used for our TH device.
So we don't offer protool to customer."

On another note: I would think that reverse engineering the protocol would not be a big thing, I just never got around to connect my USB scope yet.

peterforeman, 2017-10-26:
Same here: T/RH? found at 0x40, and no data. Only reading this AFTER i bought the sensors :-(

@cputoaster Would be great if you can get around the reverse engineering some day!

zalessky, 2017-11-21:
@bobo-t  can show on the photo where and how soldered on the Si7021? And how to configure the firmware? I have the same sensor, but I can not get it to work

bobo-t, 2017-11-21:
See below picture:
[image]
You need to disconnect the other chip (see the cut on the pcb).
Software configuration:
Module type Sonoff TH
GPIO1 None
GPIO3 None 
GPIO4 06 I2C SDA
GPIO14 05 I2C SCL

zalessky, 2017-11-24:
@bobo-t Thank you! sensor works now

ruimarinho, 2017-11-27:
@zalessky could you provide a more detailed guide on how you achieved that?

zalessky, 2017-11-27:
@ruimarinho 
1. blow out a small chip with a blowtorch
2. solder the data wires as in the picture above from @bobo-t 
3. solder the other end of them to the TN card for GIPO4 and GIPO14 - as in the picture:
http://tinkerman.cat/wp-content/uploads/2016/10/20161004_220416_LABELSs.jpg
4. You can leave the power wires for tests as is, but you can also solder them - at your discretion.
At me the sensor has earned only on firmware ESP Easy
setup as in the wiki: https://www.letscontrolit.com/wiki/index.php/SI7021/HTU21D

ruimarinho, 2017-11-27:
Thank you very much for your comment. On the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration#i2c-sensor), it says Tasmota supports SI7021. Weren't you able to get it working as well?

zalessky, 2017-11-28:
No. I could not tune in

zalessky, 2017-11-29:
@ruimarinho I found my error in the settings and this sensor earned on Tastota firmware. all OK. The sensor is not detected immediately. need a little wait :)

kkuisma, 2017-12-02:
I also bought five ITEAD's Si7021 sensors and noticed that those don't work with Tasmota. So, I took my logic analyzer and reverse engineered it. I can send more details later, if needed, but it looks their "one wire" protocol is quite simple. First, the host need to send 0.5ms low pulse, then go high for ~40us, and then move to tri-state (INPUT) to let the sensor send data back. The sensor sends totally 1bit+5 bytes/41 bits (MSB order). At first, the sensor sends a "start bit" that is "1". Next two bytes/16 bits are humidity with one decimal (i.e. 35.6% is sent as 356), two bytes/16 bits after that are temperature in Celsius, again with one decimal (i.e. 23.4C is sent as 234), and finally the last byte/8 bits is a checksum. When sensor sends logical "1" it pulls the line HIGH to ~75us, when it sends logical "0", it pulls the line HIGH to ~25us. There is about 40us (LOW) between HIGH pulses. Checksum is basically a (8-bit) sum of 4 data bytes. I also wrote a simple test code and got right values from the sensor without ITEAD's firmware, so, I believe my revers engineering is correct. @arendst, are you interested to add the support to some following release if I provide needed details?

cputoaster, 2017-12-02:
Thanks a lot! Would be a great addition!

gerritthomson, 2017-12-03:
@kkuisma well done. It would be great to have this protocol included. I bought 4 th16 & 4 si7021 prior to reading this thread.

gerritthomson, 2017-12-03:
@kkuisma looking at the easpeasy code for the dht devices the protocol looks very similar. Perhaps itead implemented the si7021 as a variant of the dht devices.

xoseperez, 2017-12-03:
I can confirm that Itead's SI7021 sensor behaves like their AM2301. They have just created a facade to make it use the same protocol. @arendst , your code is different from mine but I think you only will have to change the "delay(20);" in line 88 of the dht file to "delayMicroseconds(500);"

arendst, 2017-12-03:
Thnx both @kkuisma and @xoseperez for your input. I will release an updated xsns_06_dht.ino with the suggested change from Xose.

To make this a kind of seamless I consolidated the current DHT22 sensor selection into the AM2301 sensor selection as the handler for these two sensor has always been the same. The old DHT22 sensor selection is now changed to the new iTead SI7021 sensor using the by Xose proposed timing delay.

Give it a try and let me know your experience

kkuisma, 2017-12-03:
Excellent! Thx a lot @arendst, and @xoseperez. I just tried the latest version and with very quick and short test, it seems to work perfectly with ITEAD's Si7021 sensor :-)

jdpiguet, 2017-12-08:
So how should the module configuration look like?
I have "Module type (Sonoff Basic)" as "04 Sonoff TH" and "GPIO14 Sensor" as "03 SI7021"
But I don't get any measurement...

jdpiguet, 2017-12-08:
So how should the module configuration look like?
I have "Module type (Sonoff Basic)" as "04 Sonoff TH" and "GPIO14 Sensor" as "03 SI7021"
But I don't get any measurement...
[Edit] Problem solved: the configuration is correct, it was a HW issue!

socbrian, 2018-01-07:
Just upgraded to 5.11.0 and my si7021 works great! Thanks, probably can make this closed now

mbolsteins, 2018-02-23:
**Itead Si7021 not detected when temperature falls below +1C** 

I have a Sonoff TH10 with Itead's SI7021 temperature and humidity sensor. Flashed my Sonoff with Tasmota 5.12 and was happy that sensor is immediatley detected and measurements are displayed on webpage and published to MQTT as well.
Then I moved my Sonoff to the barn, where temperature is around 0C these days. After powering up there were no temperature/humidity readings anymore. I did some "investigation" and in syslog I could see a debug message "Timeout waiting for start signal high pulse":
```
Feb 22 16:21:51 192.168.11.107 sonoff-th-3044 ESP-WIF: Checking connection...
Feb 22 16:21:51 192.168.11.107 sonoff-th-3044 ESP-WIF: Connected
Feb 22 16:21:57 192.168.11.107 sonoff-th-3044 ESP-HTP: Main Menu
Feb 22 16:21:57 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:00 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:02 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:03 192.168.11.107 sonoff-th-3044 ESP-HTP: Configuration
```
I remember that when I tried Sonoff's original firmware I also faced some problems with temperature around 0C, but it was fine again when temperature fell below -1.

After that I checked the same sonoff with the same sensor in the house with normal room temperature and it worked, sensor was detected immediately.

aenertia, 2018-03-14:
Just tried with the itead si7021 module and brand new th16's (ver 2.1 dated 2017-6-8 on silkscreen) and dev branch of tasmota; am getting this:

 DHT: Timeout waiting for start signal high pulse

when configured as GPIO14 

Board is only powered over serial - if this matters at all, haven't tested off AC with in built rectifier. I bought a bunch of these to monitor our new house (it's an old damp house unfortunately) and the prospect of needing to mod all of them and my th16's for I2C doesn't really sound appealing.

mcoms, 2018-05-23:
Powered over AC and with a new `Sonoff TH16 v2.1 2017-6-8` and a new `Sonoff Sensor SI7021 v1.0` running `Sonoff-Tasmota 5.13.1`, this works fine.

However a second identical device and sensor gives me `DHT: Timeout waiting for start signal high pulse`. I believe this is because I plugged the sensor in with the board powered up. If you get this error you may have a damaged sensor.

closed 2018-07-10 by ascillato2

Littlesloths, 2021-04-05:
Is it possible to use more than one of sonoff's (itead) sensors with their one wire bus (not I2C) on one GPIO? I've seen others doing that via I2C, but on one-wire there seems to be no bus-scan and only one sensor possible to choose, correct?
I like to use 1x SI7021 + 1x AM2302 (to avoid using 2x same SI7021 in case that is not supported) + 1x ds18b20.

Littlesloths, 2021-04-05:
> See below picture:
> [image]
> You need to disconnect the other chip (see the cut on the pcb).
> Software configuration:
> Module type Sonoff TH
> GPIO1 None
> GPIO3 None
> GPIO4 06 I2C SDA
> GPIO14 05 I2C SCL

In case I cut the connections as per your picture, may the four pins vcc, gnd, sda, scl be used? or do I need to cut some more?