arendst/Tasmota #737: Wemos D1 mini

Gary202, 2017-08-14:
Is it possible to rum Tasmota on a Wemos d1 mini??

Gary202, 2017-08-14:
Do i flash it like normal but select wemos as the board in arduino? also are the gpios the same?

davidelang, 2017-08-14:
the wemos module in tasmota allows you to hook up any function to any gpio. I 
don't know how the D1 mini is wired, so I can't tell you what is hooked to each 
gpio

Gary202, 2017-08-14:
Thank you i will do some research and testing when i get my wemos and report back.!!

mkh595, 2017-08-14:
yes, wemos is working, here are pins you can use

[image]

here is example with 2x DS18B20, 1x DHT22, 1x BME 280 and 3x Relay
[image]

closed 2017-08-15 by Gary202