arendst/Tasmota #738: Adding second relay has not enough power

Krettoss, 2017-08-14:
Hi,
I am using sonoff basic and I have problem to add second relay as 18 RELAY2 on GPIO14. I can measure 3V on GPIO14 but it is not enough to switch relay. I bought also low signal relays and still dont have enough juice. The main relay has higher power consumption and it works. Is there any differens in GPIO pins? Is another pin stronger ? GPIO1, GPIO3, GPIO4 ?

davidelang, 2017-08-15:
all gpio pins are 3.3v, if your relay requires a higher voltage, then you need 
to put a transister between them to control the relay.

ionciubotaru, 2017-08-15:
try to locate the 5V circuit on the PCB, it supplies power to the first relay.
Use a relay like [this](https://www.banggood.com/DC-5V-1CH-Relay-Shield-V2-Version-2-For-WEMOS-D1-Mini-ESP8266-WiFi-Module-p-1102379.html?rmmds=search)

jetema, 2017-08-17:
The relay on the sonoff basic works with 5V, wich is the main voltage on the power suply, there's a regulator U2 that converts from 5V to 3,3V to supply ESP8266 and Flash memory, look at the picture
[image]
You'll find the 5V on D5 terminals,  ;-)  good luck Krettoss !

Krettoss, 2017-08-17:
Thanks for all replays, I have found 5V for relay. Do I have to use transitor on pin GPIO14 to send 5V to relay? Or is there any other way to control 5V for relay? My relay is SRD-03VDC-SL-C.

jetema, 2017-08-17:
It is always suitable using a transistor between a digital pin and a relay, just have a look at the original connection on sonoff with the 5V relay :
https://goo.gl/photos/BzZDt3hAGG5NpkrJ9
Your relay works with 3V, so you may connect same way with a resistor to decrease voltage on coil of the relay

Krettoss, 2017-08-17:
Can I use 3,3V pin for my 3V relay instead of 5V pin ? Or which resistor should I use? I prefer solution without resistor. Thanks

jetema, 2017-08-18:
You could use this way
[image]
Never connect a relay direct to a gpio, otherwise will be damaged

closed 2018-05-07 by stale[bot]

easyyu, 2022-10-19:
> You could use this way [image] [ [image] ](https://user-images.githubusercontent.com/30664452/29453269-51b224ac-8409-11e7-9fdd-985beabafe19.gif) [ [...]

Can you please advise which transistor did you use?
And which GPIO is most reliable?