arendst/Tasmota #741: S20 extra header pins? - Help wanted.

jat80, 2017-08-16:
[image]

I've used the bottom ones to flash with tasmota but wondering if I wanted to add two sensors to this if I can use the other pins?

So bottom pinout will use gpio 3 (serial in) for temp/humidity sensor and side pinout using gpio 1 (serial out) for motion sensor along with the  corresponding 3.3v and ground pins.

eggfat, 2017-08-16:
Hi,
Sorry I cant help you with with the sensors question.
I do have a question. Have you used blue tack / putty to attach the 4 pin header?
if so did it work?

Thanks,
Martyn.

ionciubotaru, 2017-08-16:
Yes you can use. The pins are TX and RX and in S20 are used only for programming reasons
Disable serial log and select the sensor you need

ionciubotaru, 2017-08-16:
@eggfat you can attach a FTDI to the 4 pin header and open the serial port using putty at speed 115200

eggfat, 2017-08-16:
@ionciubotaru Thanks for the tip. @jat80 @ionciubotaru What kind of sensors could be used in the S20?

ionciubotaru, 2017-08-16:
Any type of sensors which you can connect to gpio1 and gpio3 and suported by Tasmota
  "DHT11",
  "AM2301",
  "DHT22",
  "DS18x20",
  "WS2812",
  "IRremote",
  "Switch1",
  "Switch2",
  "Switch3",
  "Switch4",
  "Button1",
  "Button2",
  "Button3",
  "Button4",
  "Relay1",
  "Relay2",
  "Relay3",
  "Relay4",
  "Relay1I",
  "Relay2I",
  "Relay3I",
  "Relay4I",
  "Led1",
  "Led2",
  "Led3",
  "Led4",
  "Led1I",
  "Led2I",
  "Led3I",
  "Led4I",
  "PWM1",
  "PWM2",
  "PWM3",
  "PWM4",
  "PWM5",
  "Counter1",
  "Counter2",
  "Counter3",
  "Counter4"

jat80, 2017-08-16:
@ionciubitaru just to confirm you are saying I can use the pins?  Also what do u mean by disabling the serial log?  Is this only if I'm adding a sensor to those 2nd sets of pins?

jat80, 2017-08-16:
@eggfat I initially just held them in with bluetac on an angle to flash with the firmware.  I've since soldered in some pins to add a sensor.  This version of the S20 (AU/NZ) doesn't allow u to get to the underside of the board so it's a little tricky soldering the pins.  I'll post a pic later if how I did it.

mattwelly, 2017-10-01:
@jat80 any chance you could share how you soldered the pins? mine powers the unit up after soldering but no tx/rx I'm afraid.

jat80, 2017-10-01:
@mattwelly As per the pic above,  I just made sure I held the them on an angle so the actual pins with touching the side of the hole when flashing the unit using the bottom pins NOT the ones on the side.

As for soldering them. I removed the black plastic from the header pins to leave just the metal pin itself.  I then put solder into each hole to fill it.  I then heated the solder up again and when it's liquid I inserted the metal pin holding it in place until the solder then goes hard again.  Repeating for each pin hole.  

Hopefully that makes sense but it was the only way for me to get permanent header pins on the AU/NZ version of the S20 as you cannot get to the underside of the board.

No rx/tx is probably because of:

1) you have not soldered them on properly for a good connection or
2) you need to swap the  rx/tx connections around depending on what you are using to connect the unit to your computer (FTDI, Arduino etc) to flash.

Hope that helps.

TommySharpNZ, 2017-11-18:
Hi did you get your extra sensors connected to this?
I'm trying to create a "smart night light" so would need temp, motion, light and LED so need 4.. But I don't think it will be possible.

closed 2018-05-07 by stale[bot]