arendst/Tasmota #742: sonoff pow no voltage reading

simondid, 2017-08-16:
hey

so I'm setting up my first sonoff pow and can't get it to make a voltage reading :( it measures current / power totally fine but no voltage 

system info

Program version | 5.5.2b
Build Date & Time | 2017-08-15T19:13:40
Core/SDK version | 2_3_0/1.5.3(aec24ac9)
ESP Chip id | 11837725
Flash Chip id | 1458400
Flash size | 4096kB
Program flash size | 1024kB
Program size | 465kB
Free program space | 472kB
Free memory | 24kB

backup configuration file if this can help in any way
[https://www.file-upload.com/twvqwmb4mtbd
](url)

any suggestions why this is?

arendst, 2017-08-17:
Did it work with the original firmware?

Voltage  or current are alternated provided by a single pin on the hlw chip. If you do not receive voltage readings it might mean the chip is broken.

simondid, 2017-08-17:
i have trayed using both commands and still nothing :(

as for the original firmware, i can't remember if it showed voltage, but correct me if I'm wrong but in order to show power/watts it needs the current and voltage readings to calculate power/watts as it can't measure power/watts?

arendst, 2017-08-17:
Alas, the hlw chip calculates power and provides it on a pin different from the pin that  alternated provides either voltage or current.

I'm afraid you hlw chip is broken.

arendst, 2017-08-18:
There might also be a problem with soldering the chip to the board.

[image]

If pin 8 is not soldered correctly it fails to switch between voltage and current readings. You might want to check if the chip is soldered correctly. This pin should be connected to the esp8266 pin 24 (GPIO05)

simondid, 2017-08-18:
so i went over all the solder joints and everything seems to we soldered fine :(  but as you say as the power comes from pin 6 and current / voltage comes from pin 7 and 8 there might be a selection problem and therefore it defaults to current for unknown reasons :( but fortunately I only need the power output any ways :P 

I'm no electrical professional but let's say i started measuring on pin 8 of the HLW8012 chip shut I expect a 3.3v / 0 v selection signal? with reference to the esp ground?

arendst, 2017-08-18:
Correct. It might toggle every 1.6 second so must be easily measurable.

simondid, 2017-08-18:
so I did some measuring, and found that pin 8 changes between 5v and 0v so it seems that the chip is getting the control signal :P  

any other suggestions?

arendst, 2017-08-18:
Try to measure something on pin 7 of hlw during both pin 8 high and low.

After that I'm out of options

simondid, 2017-08-18:
when measuring between pin5 and pin7 i'm getting 3.08v  but after some more measuring i was unstable to  recreate the selection signal :( but i fund a wired solder joint on the board 

[http://imgur.com/a/gRPGG](url)

is this normal?

closed 2018-05-07 by stale[bot]