arendst/Tasmota #743: Can I control DC voltage with sonoff?

huseyint, 2017-08-17:
This is not mainly related with tasmota but I want to know whether I can control 12V DC voltage with sonoff? I have led lights over my kitchen counter which are powered by a 220V AC -> 12V DC adapter, I want to control those lights with sonoff.

jpenninkhof, 2017-08-17:
It would be best to put the sonoff between the wall socket and your adapter rather that between your adapter and the led strip. The sonoff won't work on 12V DC.

huseyint, 2017-08-17:
Oh right, I only though the switching (relay) capabilities for DC voltage and never thought the powering side of things. Thanks.

closed 2017-08-17 by huseyint

davidelang, 2017-08-17:
itead does sell DC 12v modules, so you can buy those and they work well, just 
avoid the 'inching' one.