arendst/Tasmota #744: Potential Tasmota restart/reconnect problem when using Home Assistant

diplix, 2017-08-17:
(This is a wicked problem I experienced with Tasmota and Home Assistant, where, from time to time, one of my switches turned on at night. I (think) what’s below solves the problem, before I send a pull request for the Wiki page, I’d like to discuss it here first or ask if anyone else experienced this.)

If you configure a Sonoff switch with Tasmota in Home Assistant (HA) like it is describe [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant) and additionally use GPIO 14 as a (toggle) switch, there’s a potential problem when the Sonoff loses it’s connection and reconnects or restarts.

When you tell HA to retain the mqtt values, it flags both `stat/sonofftwo/POWER` and `cmnd/sonofftwo/power` topics as retained. 

So if HA turns the Sonoff on, both `stat` and `cmnd` topic are set to `ON` (and retained).
If you switch off the Sonoff with a (pysical) switch attached to GPIO 14, the `stat` topic gets updated by Tasmota — but not the `cmnd` topic!
Usually this is not a problem, since HA only cares about the `stat` topic and ignores the `cmnd` topic. So does the Sonoff, as long as the `cmnd` topic doesn’t change, it’s divergent status will be ignored.

However, if the Sonoff loses it’s connection or restarts, it checks the `cmnd` topic. In the scenario described above, it is set and retained to `ON`, although we turned the Sonoff `OFF` with the hardware switch attached to GPIO 14. So after the Sonoff regains online status, it follows the retained `cmnd` topic and turns `ON` the switch.

This is how i worked around it:

Setting `SwitchMode`, `SwitchTopic` and `SwitchRetain` to `1` lets the Sonoff send a retained `cmnd` topic each time the pysical switch is toggled. The switch topic is different to the default `cmnd` topic, though: `cmnd/sonofftwo/POWER1`

So if you configure Tasmota like this

```
SwitchMode 1
SwitchTopic 1
SwitchRetain 1
```
and HA to the `cmnd/sonofftwo/POWER1` topic, everything should be in sync again.

```
switch:
  platform: mqtt
  name: "sonofftwo"
  state_topic: "stat/sonofftwo/POWER"
  command_topic: "cmnd/sonofftwo/POWER1"
  qos: 0
  optimistic: false
  retain: true
```  

You might also need to delete the retained values in `cmnd/sonofftwo/power`, as described [here](https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration) (`mosquitto_pub -p 1883 -u <username> -P <password> -d -n -r -t cmnd/sonofftwo/power`), otherwise the Sonoff might continue to consume the retained value in that topic after a restart or reconnect.

davidelang, 2017-08-17:
to clarify, the retain flag isn't telling HA to save the last message, it's 
telling the MQTT server to retain the last setting.

Also, Tasmota does not 'check' for the last command at startup, it just connects 
to the MQTT server and tells it that tasmota wants to subscribe to the 
cmnd/topic/* messages

at that point, the MQTT server issues the retained command, and at that point 
Tasmota has no way of knowing that these are commands retained by the MQTT 
server as opposed to new commands being sent from the HA server

Setting the switchtopic will decouple the switch from local actions, and yes, 
the mqtt server can then retain those commands. But the Tasmota device will not 
respond to this second topic, it only sends with that topic.

What you really want it to not have anything retained and let poweronstate keep 
the tasmota in the state it was in before the restart/reconnect, but then you 
have a problem with HA becuse it doesn't know how to watch for tele/ messages.

a possible work-around for this would be to have tasmota send a stat/ message 
when it uses poweronstate so that HA will get a message of what state it is 
actually in.

diplix, 2017-08-17:
Tasmota does react to the topic `cmnd/sonofftwo/POWER1`, so if you meant that as the second topic that Tasmota doesn’t respond to: it does, and the workaround I described works (for me).

Another workaround would be to instruct  HA to set the `cmnd` topic accordingly, each time it receives a `stat` message. but for that I would have had to set up an extra automation.

joecotton, 2017-08-17:
> a possible work-around for this would be to have tasmota send a stat/ message
when it uses poweronstate so that HA will get a message of what state it is
actually in.

This seems reasonable. But doesn't it already do this? I just restarted one of mine (firmware 5.5.1) and right after the startup MQTT messages, there is a stat message with the current power state.

davidelang, 2017-08-17:
On Thu, 17 Aug 2017, joecotton wrote:

> This seems reasonable. But doesn't it already do this? I just restarted one of mine (firmware 5.5.1) and right after the startup MQTT messages, there is a stat message with the current power state.

If tasmota does send a stat message on startup, then the right thing to do here 
is to disable the retain setting on tasmota and HA, and clear the retain state 
from the MQTT server.

At that point the tasmota poweronstate can work and HA will know the state of 
the device at all times.

cryptooth, 2017-09-07:
Have you solved this problem completely? How is it solved? There is a similar problem with my Sonoff Touch. I think MQTT is disconnecting / reconnecting. When I close it by touching it, it turns on by itself and the lamp stays on all night. But there is no problem if I close the lamp via HA or script.

davidelang, 2017-09-07:
read the page on poweronsetting, and specifically the section about the mqtt 
server retaining (and replaying) commands.

MastroPino, 2018-02-12:
This saved me, I come hear ‘cause Google Yelp me tuo find this post. On the tasmota wiki there is all scenario but is not easy to find for all

asantos15, 2018-02-21:
Tried @diplix `s approach, but without success.
Would there be anything additional to the steps explained above? The garage door device keeps turning on and off at pi`s reboot and it`s really not acceptable.
Thanks a lot in advance for sharing!

zainuluk, 2018-03-26:
I also have the Tasmota Sonoff lights turning on randomly during some point between 11pm and 2am. I assume it is because it gets disconnected from the router but checking the logs I can't see any mention of this. Is there a way to get some information when the device disconnects from the router?

asantos15, 2018-03-26:
@zainuluk , you should look at then how far it is from the router. I used to have this disconnect issue, because I had a paper sheet on the top of my relay to avoid anyone touching it, but it kept disconnecting from the router.
If this is disconnecting at boot, or when connecting back to MQTT, then you should try the following:
This is the command that publishes nothing and overrides the state that is causing the board to change state:
On the terminal, to check if there is any state pre selected: 
mosquitto_sub -p 1883 -u <username> -P <password> -t cmnd/+/power changing username and password only. Then...
mosquitto_pub -h hostipaddress -p 1883 -r -t 'cmnd/"devicename"/POWER' -n -u username-P password

zainuluk, 2018-04-06:
@asantos15, the question is, why does it only happen at night and not the rest of the day. Some nights it will not happen at all and then other nights it will happen 4 or 5 times. I moved the lamp closer to the router, less than 1 meter away (previously it was about 8 meters away) and the light still turns on randomly.

asantos15, 2018-04-06:
It looks like it's not synchronized with HA config wise then. I will post my configuration once I get home tonight - 12 hours from now.

MastroPino, 2018-04-06:
I find a solution form my problem and its described here: https://community.home-assistant.io/t/mqtt-and-tasmota-problems-when-disconnect-crash/47016

joealmonte, 2018-05-23:
When Tasmota senses a Wi-Fi disconnect, the BlinkCount counter initiates. By default, the rate for this counter is set to 1 blink per second. Yes, a Wi-Fi disconnect is one of the instances when your Sonoff's green LED starts flashing.  After 10 seconds elapse, Tasmota will trigger the Sonoff to restart. It will continue to do this until it re-establishes a connection with the Wi-Fi. Once it reconnects, Tasmota initiates its safe mode System Restart procedure:

00:53:35 MQT: tele/sonoff5/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_7C6F6F","GroupTopic":"sonoffs"}

00:53:35 MQT: tele/sonoff5/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff5-3951","IPAddress":"192.168.0.205"}

00:53:35 MQT: tele/sonoff5/INFO3 = {"RestartReason":"Software/System restart"}

...causing the garage door to open without consent. To minimize these events from happening, the BlinkCount can be increased. 

I own a Netgear R8000P router, and every time I manually reboot the router, it will trigger my garage door to open. The router I own takes over to minutes to power cycle. In my case, the garage door Sonoff 
restarts about 15 times before reconnecting to Wi-Fi. So to workaround the issue, I had to increase the BlinkCount before power restore to 180 (3 minutes).

01:02:44 CMD: blinkcount
01:02:44 MQT: stat/sonoff5/RESULT = {"BlinkCount":10}
01:05:19 CMD: blinktime
01:05:19 MQT: stat/sonoff5/RESULT = {"BlinkTime":10}
01:09:06 CMD: blinkcount 180
01:09:06 MQT: stat/sonoff5/RESULT = {"BlinkCount":180}

This configuration now gives enough time for the router to restart, and the Sonoff to re-connect to WiFi without triggering a power restore condition, or restarting. I hope this helps a few of you guys!!!

ascillato2, 2018-06-27:
Hi, if your issue is not solved, please, reopen it. Thanks.

closed 2018-06-27 by ascillato2