arendst/Tasmota #746: Is it possible to add result of "AnalogInput0":xx to the web page ?

roblad, 2017-08-17:
Hi,

I am requesting again.

Is it possible to add result of "AnalogInput0":xx to the web page ?
Nice to have also some settings for ADC to set USE_ADC_VCC or AnalogInput0 visable on the web page.

Additionally I suggest function for LDR that could be easier then use LX measurement through I2C light sensitivity
It is possible by monitoring ADC.

Thanks,

arendst, 2017-08-17:
I will add adc0 to webpage.

Switch option is impossible as it only functions after a recompile. See arduino-esp for background.

roblad, 2017-08-17:
thanks, but could you add V and when it does not exists only number value.

roblad, 2017-08-18:
Hi 

Thanks for it, but I changed code in webserver.ino lines 497-509 to this
#ifndef USE_ADC_VCC
  if (pin[GPIO_ADC0] < 99) {
    snprintf_P(svalue, sizeof(svalue), PSTR("<tr><th>AnalogInput0</th><td>%d</td></tr>"), getAdc0());
    tpage += svalue;
  }
#else
if (pin[GPIO_ADC0] < 99) {
  char stemp[TOPSZ];
  dtostrf((double)ESP.getVcc()/1000, 1, 3, stemp1);
  snprintf_P(svalue, sizeof(svalue), PSTR("<tr><th>VCC:</th><td>%s V</td></tr>"),stemp);
  tpage += svalue;
}
#endif
it shows VCC, do you think that it will be usable for any body ?

arendst, 2017-08-18:
No, I would not wanting to see non-info on a normal sonoff webpage. Just keep it as your local code.

closed 2018-05-07 by stale[bot]