arendst/Tasmota #750: Strange behaviour on PREFIX2

stefanbode, 2017-08-17:
I have changed the PUB_PREFIX and PUB_PREFIX2 to the same value "hm/status".

If I initiate "status 0" from the console the messages get correctly send to "hm/status/...." as expected. 

If I send through MQTT the command "hm/setting/status" "0" I also get the list in the console, but this time the messages are send to "stat/....." I thought this was the same procedure but behaves differently. 

In console i got for prefix1 to 3 the correct assignments. Any idea?

arendst, 2017-08-17:
Apparently you are using the fallback topic with mqtt. Try your topic and the result will be as your console input.

stefanbode, 2017-08-17:
yes looking into getTopic_P and had the same idea that there is a fallback or something like this. Can you please explain what you meant with "try yourtopic". In 5.2.4 I send for example through MQTT "hm/setting/ESP_4711AS/status" "0" and everything was fine. Now the same command delivers the fallback. What is wrong with the configuration or what did I missed?

Topic is changed to the device name ESP_4711AS and Full Topic is unchanged.

stefanbode, 2017-08-17:
Ok, found. Too much copy and paste. The problem was I named the Device like the Topic instead of DVES_... like normal. Thanks for your support

closed 2017-08-17 by stefanbode

arendst, 2017-08-17:
Since 5.2.3 there is a major difference between fallbacktopic and topic. topic uses fulltopic too and is also used by console. fallbacktopic only uses cmnd/... and does not use fulltopic.

Make sure topic is unequal to fallback topic.