arendst/Tasmota #751: Sonoff T1 2CH 2nd channel not working

cputoaster, 2017-08-18:
in 5.6.0, with T1 2CH, when configuring module as T1 2CH, the button "Toggle 2" is not triggering any relay action. When configuring it as T1 3CH, button "Toggle 2" triggers the 2nd relay just fine.

arendst, 2017-08-18:
Thnx for the input. I had to deduce it's functionality from a 3CH type. 

Pls perform the below changes in file sonoff_template.h starting at line 545:
```
  { "Sonoff T1 2CH",   // Sonoff T1 2CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     0,
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
```
and let me know if it solves your problem.

cputoaster, 2017-08-18:
yep, works.

closed 2017-08-18 by arendst