arendst/Tasmota #753: WeMos D1 mini Module 433MHz & IR LED

zibous, 2017-08-19:
**433MHz Wireless Transmitter MX-FS-03V**

[image]

**Diode LED IR Emission used as IRServer with MQTT**

Is it possible to use these two components for SONOFF WeMos D1 mini modules? 
I would like to switch devices with MQTT commands with the LED IR or 
with the Wireless Transmitter MX-FS-03V

arendst, 2017-08-19:
There is no (panned) support for 433MHz in Tasmota other than the Sonoff Bridge which does manage 433 transceiving and detection.

Infrared is implemented using the IRRemoteEsp8266 library.

zibous, 2017-08-19:
Too bad for me... :) , so I have to look for another solution, 
but thanks for the super performance for this software.

closed 2017-08-19 by zibous

nosaj66au, 2018-01-04:
Hi Zibous,
  Yes got this working tonight. by add a protocol to irremote but driving a 433 transmiter. Its now controlling my blinds. :)

Jason

profesionalTI, 2018-01-04:
Hi, I want to control May TV with IR, please can you tell me the string and step 

Coordialmente,
Best Regards,

Marcelo A. Anuch Medina
Consultor de Empresas Público/Privadas
Ingeniero Comercial, MBA UDD
Business Management

🔒   La información contenida en esta transmisión es confidencial y está dirigida exclusivamente a el o los destinatarios indicados. Cualquier uso por otras personas que no sean sus destinatarios está sancionado por la ley chilena. Si ha recibido esta transmisión por error, por favor destrúyala y notifique al remitente si corresponde. Las opiniones expresadas en este correo son las de su autor y pueden no ser compartidas por la compañía. Marcelo A. Anuch Medina,  no asume obligaciones ni responsabilidades por el contenido de este correo. Tampoco garantiza que la transmisión de este correo sea segura o libre de errores, y en consecuencia, no responde por virus, pérdidas, destrucción, recepción tardía o incompleta.

🍀  Antes de imprimir este correo electrónico asegúrese que sea realmente necesario: Cuidemos el medio ambiente! Todos vivimos en un solo planeta.