arendst/Tasmota #754: WeMos-Tasmota development

decologne, 2017-08-19:
Hi Theo,
Is it interesting to develop an image WeMos-Tasmota based on Sonoff-Tasmota ?
The WeMos image include then sensor modules, oled displays etc. with 4G memory.
What do you think about this ?
Regards, Jan

arendst, 2017-08-19:
It's a nice idea and a major task. I won't keep you from trying ;-)

mihalski, 2017-08-19:
I'm sure many users would love to have such a branch. I have more Wemos units running Tasmota than I do Sonoffs. I'd happily contribute to testing.

I guess ideally you'd want to minimise any deviation so that you could stay in sync with the Sonoff-Tasmota development without too much trouble.

decologne, 2017-08-19:
Theo, can I call you for consultation in Dutch ? If yes, mail your tel. number.
Regards, Eau

closed 2017-08-23 by decologne